procount/pinn #208: Backup Failed

darkplatinum, 2018-12-19:
My backup failed. I am using a 1TB seagate hdd paired with a 32GB sandisk ultra Micro SD card. I was backing up 2 Raspbians as well as 1 retropie. It said check debug log for errors. Where is the log located at?

procount, 2018-12-19:
When you say you "paired" the 32GB micro Sd car dwith the 1TB HDD, do you mean that you have PINN on the SD card and you installed the OSes to USB? So you have the boot partitions on the Sd card and the rootfs partitions on the HDD?

Please see the [Trouble shooting](https://github.com/procount/pinn/wiki/Troubleshooting) page of the wiki for details on how to see the debug log. Point 6 provides particularly useful information.

Maybe next time try backing up your OSes one at a time to identify which one of them is causing a problem.

procount, 2018-12-20:
> Hmm... I'll try doing the backups separately.

That might be a good idea, especially for the first time, til you see how it works, and to make sure it works for your particular set of circumstances. Once you know they all work and you can restore them all, you could perhaps do them altogether.

Did you follow the advice in the troubleshooting page of the wiki to download the savelog script and put it on your SD card before you backup? It will make debugging easier if things go wrong. The debug log is written to the temp file system in RAM, so it is lost whenever you reboot the RPi. So having the script prepared will be a good idea so you can capture the error logs if the problem occurs again.

> Would the backup work if the date changed?

Yes, of course. The date is only used at the beginning of the backup to create a name for this new backup so that you can tell them apart. When dealing with backups and trying to restore them, the date is probably the most important thing, which is why I add it to the OS name. You can override it with anything you like if you prefer, but I suggest you stick with the default for now.

darkplatinum, 2018-12-20:
No, I had not thought about using the script, but now that you mention it I will. How do I access the recovery partition while I am booted into Raspbian? What folder is it in /media/pi?

darkplatinum, 2018-12-20:
Ok done. So if I receive an error, I immediately press Ctrl-Alt-f2, (while in PINN) then enter my login details (I changed the root password).

Do I then type /mnt/pinn or /mnt/savelog?

procount, 2018-12-20:
You may have changed the root password in Raspbian, but you can't change it in PINN - they are separate. So you should use root/raspberry. and type `/mnt/savelog` as written in the troubleshooting section.

Now, as you asked previously about how to copy the savelog using Raspbian, I am assuming you are using Raspbian as your main OS, rather than using a Windows laptop or something....?

So I'm assuming once you have captured the log, you're going to want to know how to access it from Raspbian? In this case, boot into Raspbian and use the following commands:

```
mkdir -p /tmp/pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cp /tmp/pinn/log  /home/pi
sync
sudo umount /tmp/pinn
```
Then the `log` file will be in your Raspbian home directory and you can upload it and add it to this issue.

darkplatinum, 2018-12-21:
Had to rename the file with the extension .txt as github did not want it to be uploaded without a format.

[log.txt](https://github.com/procount/pinn/files/2702144/log.txt)

procount, 2018-12-21:
I will try to create the same scenario.

I assume you have plenty of space on your backup usb stick? I think you said it was 64gb or something so it should have space.

procount, 2018-12-21:
I managed to replicate some of your issue.
There seems to be a problem calculating the disk space required for the backup when the OS has been installed to USB. Despite the size being incorrect, it does seem to backup and restore correctly, though.

I have not managed to replicate the backup button not doing anything, though. If you manage to formulate some repeatable steps, please let me know and I will try to repeat it.

darkplatinum, 2018-12-21:
Hmm. So you are saying even though it says an error, the files should have been backed up? I have 2 folder on my USB drive (backup drive) both raspbian, but the second raspbian has a root.tar.gz of 4 gigs whereas the other has 178.6mb

procount, 2018-12-21:
Mine did not error and I could restore it.
I suspect if one of yours is only 178mb that it failes whereas the ither might have succeeded. You can only know by trying to restore it to another drive.

procount, 2018-12-22:
Please download the following zip file:
https://sourceforge.net/projects/pinn/files/testing/301a.zip/download
Unzip it and you will find `recovery.rfs` inside it.
Copy recovery.rfs to PINN's recovery partition and overwrite the file that is already there.
Reboot PINN and try another backup. 
I hope it should now get the partition sizes correct when an OS is stored to USB.

procount, 2018-12-22:
Please download the following file for v3.0.1a
https://sourceforge.net/projects/pinn/files/testing/301a.zip/download
Unzip the file and you will find `recovery.rfs` inside it.
Copy this file to PINN's recovery partition and overwrite the existing file.
Sync and reboot.

Hopefully you will find the sizes are correctly calculated and the progress bar updated for when the OS is stored on a USB drive

darkplatinum, 2018-12-22:
Could you provide instructions on how to copy it via raspbian? Don't want to mess it up.

procount, 2018-12-22:
Maybe just wait a bit, because I tested it and it seems ok so I'm going to release it.

procount, 2018-12-22:
v3.0.2 is now released.

darkplatinum, 2018-12-26:
Tried the new version today, it still gave problems.
[log2.txt](https://github.com/procount/pinn/files/2709838/log2.txt)

procount, 2018-12-26:
Hmm. That's odd. I will investigate

procount, 2018-12-26:
How quickly does it fail, in terms of time or amount of data backed up?

darkplatinum, 2018-12-26:
Sorry, not sure. Don't really have the time to look at it. I started it around 6:40am this morning. Came back around 8:30am and it was like that.

procount, 2018-12-26:
No worries. I'll try it myself anyway when I have some time.

darkplatinum, 2018-12-26:
In the mean time just enjoy your Christmas holiday. This is not as important :smiley:

procount, 2018-12-27:
Please try v3.0.3 which is now uploaded, so it should upgrade automatically.

darkplatinum, 2018-12-29:
Sorry, haven't been here for some time. I'll try to do it tonight.

darkplatinum, 2018-12-31:
Unfortunately, I have forgotten to do so. Ill make sure to do it within 2 weeks

darkplatinum, 2019-01-02:
Ok did it again today. Updated PINN. Still getting errors. The error happen some time after 2500mb was done and before 50minutes was up.

procount, 2019-03-15:
Are you still having this problem? I changed the backup routine a bit in the latest v3.1.1, so I don't know if that will improve the situation for you.

darkplatinum, 2019-04-01:
I haven't tried backing up since the issue. I will try again but I don't think it will be any time soon as I am busy these coming weeks.

procount, 2020-04-02:
No activity, so closing. Please re-open if necessary.

closed 2020-04-02 by procount