procount/pinn #212: Please update Gentoo64 images (v1.3.1 released)

sakaki-, 2018-12-28:
Hi @procount -

I've just released **v1.3.1** of the [gentoo-on-rpi3-64bit](https://github.com/sakaki-/gentoo-on-rpi3-64bit) image (including Pi-Top variant), and redone the first slide of the marketing deck to match (available (as before) [here](https://github.com/sakaki-/resources/blob/master/raspberrypi/pi3/noobs/Slide1.png)).

Whenever you get a chance, would you mind updating the images on PINN? Please note that with this image, the minimum size of target SD card has increased to 16GB (up from 8GB), due to the Chromium browser now being bundled.

> Well, technically the initial (uncompressed) image size is now 11,328MiB (of which the root partition is 11,264MiB (11GiB), prior to autoexpand). 11GiB for the root partition is a reasonable 'real minimum', if you want a 1GiB swapfile, and some space to run `emerge`s in etc.

Many thanks for your help!

sakaki

procount, 2018-12-29:
I'm in the process of converting it now. the root partition seems to have only about 7GiB of data, so you need me to add another 4GiB of free space as a minimum requirement, making it 11GiB in total?

procount, 2018-12-29:
@sakaki- Do you know if this is compatible with the 3A+? I marked it as such, but I haven't got one so I wasn't sure, due to it having less memory.

procount, 2018-12-29:
I modified the partition_setup.sh script to make it compatible with PINN's new OS backup facility.
v1.3.1 is now released.

sakaki-, 2018-12-29:
@procount,

As to the 3A+, I don't have one of these either, currently, to test on. Does it require anything different to the 3B/B+ in terms of dtbs or kernel naming for boot? In any case, 512MiB is tight to be running `vc4-fkms-v3d` with a 256MiB CMA, which is the shipped default, and an Xfce4 desktop... ><

For the root partition, yes, please set 11GiB as the minimum root requirement. There's no point people installing a system if it fails out on the first automatic weekly update, which has happened to a few people who wrote 1.3.0 onto 8GB cards.

Thanks for releasing this so quickly!

Best, sakaki

closed 2018-12-29 by sakaki-