procount/pinn #214: Unsupported WiFi Password Characters

JeffDann, 2018-12-30:
I was attempting to connect to a WiFi network that had an '@' in the password but it would not accept it. I finally had to temporarily change the password in order to connect to WiFi.

procount, 2018-12-30:
This could be a keyboard layout/language problem. Did you try checking the "show password" box first to check that the characters you think you are entering are actually being used? For example, on some keyboards, the quote (") and the @ symbol are often interchanged.

Alternatively, you can copy a working wpa_supplicant.conf file to PINN's recovery partition and PINN will use that. See https://github.com/procount/pinn/blob/master/README_PINN.md#preconfiguring-a-wifi-network

procount, 2019-03-15:
Presuming resolved.

closed 2019-03-15 by procount