procount/pinn #216: wpa_supplicant file won't connect to WiFi

ajtwillrule, 2019-01-05:
I use the wpa_supplicant file to make a headless install of pinn but it doesn't seem to connect, and then when I connect it to a monitor and manually try to connect to WiFi, it says "No wifi interfaces available" when I am using a RPI 3B. If you need me to test things differently just ask.

procount, 2019-01-05:
There should be no problem using the onboard wifi.
Does the wifi work ok with plain Raspbian installed by Etcher?
If so, copy the wpa_supplicant.conc file from there and paste ot into PINN to ensure a known working file.

ajtwillrule, 2019-03-03:
No, it didn’t work with plain raspian either, sorry I didn’t respond earlier

lurch, 2019-03-03:
Well the first thing to do then is to get plain Raspbian connecting https://www.raspberrypi.org/documentation/configuration/wireless/README.md and then you _should_ be able to get PINN working with the same config file.

procount, 2019-03-15:
No activity. Looks like wpa_supplicant configuration or hardware problem.

closed 2019-03-15 by procount

ModernShrub, 2021-06-24:
hi

ModernShrub, 2021-06-24:
You should copy paste this, it worked for me:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
country=<Insert 2 letter ISO 3166-1 country code here>
update_config=1

network={
 ssid="<Name of your wireless LAN>"
 psk="<Password for your wireless LAN>"
}