procount/pinn #217: Problem installing PiNN

GabrielBalan90, 2019-01-08:
I have tried everything that I thought and searched on Google. Hope you guys can share some light.
Bought a new Samsung Evo+ SD card to install PiNN-lite, formated as FAT32 and now it gets stuck everytime to this stage which can be seen in the images attached.

I only got to the installer to the the step in which I had to select a "Project Space" and from that point on I had to connect it to Wi-Fi, I have used the HotSpot on my phone for this because my keyboard is wireless so no possibility of connecting to the WI-FI without a password afterwards it was gone - EMMERGENCY SHELL.

SD card has been formated with Gpart, fdisk and Etcher (which now has a 'bug' and detects the SD as if it already has an image on it even if the SD is completely empty) and it still gets stuck to the emergency shell.

Has anyone ever encountered this problem in the past or can share some light on the issue?

Image 1: https://imgur.com/a/yUZBlA3
Image 2: https://imgur.com/a/jD7oK1Z

procount, 2019-01-08:
It sounds like you've managed to format your SD card without an MBR.
How big is your SD card? If it is larger than 32GB, you will need to use special tools to format it as FAT32 otherwise it will end up as exFAT.
Do you have a Linux PC or Windows PC that you format the SD card on?
If you have Windows, use the SD association's SD formatter tool to format the card. Digital Cameras are also a good alternative.
If you are using Linux, make sure you create a partition before formatting it as FAT32.

Normally I would suggest downloading the Etcher image version of PINN and using Etcher to flash it to the SD card, but I'm not sure why Etcher is having an issue with your card. (You can't use Etcher to flash pinn-lite.zip - you must flash pinn-lite.img.zip - see PINN's documentation)

lurch, 2019-01-08:
I wonder if it might be a not-powerful-enough PSU for the Pi, and so when you start up the WiFi on the Pi it causes the SD card to become inaccessible? That's purely a guess though :man_shrugging:

darkplatinum, 2019-01-09:
Why not try using this: [Pinn sourceforge .zip](https://sourceforge.net/projects/pinn/files/pinn-lite.zip/download)

Extract the PINN files, drag and drop to the SD card.

GabrielBalan90, 2019-01-09:
Darkplatinum, did that, as I said, I have read the documentation.

procount, 2019-01-09:
It might be worth checking your SD card with h2testw, just to rule out any problems with it.

> Procount, the SD card is 32gb and it has been formatted using Windows 7, same problem as in the images, formatted as FAT32 everytime.

But did you use SD formatter, or Windows internal tools? SD formatter will also wipe out any previous partitions.

> I used Etcher and downloaded pinn-lite-img but I get this bug and the SD is completely empty, no image on it: https://imgur.com/iT1GMlX 

It looks like you are using Etcher on Linux, but the drive selected is /dev/sda, which is probably your hard disk and NOT what you want. You need to select your USB drive, which could be /dev/sdb or something similar.

Another alternative to try is, if you can get plain Raspbian onto your SD card and working, just replace the boot partition with the contents of pinn-lite.zip and see if that will fix it (although it is a very similar approach to using Etcher with pinn-lite.img.zip)

GabrielBalan90, 2019-01-09:
Procount, I will use SD formatter tonight and give it another try. I used Windows internal formatter.
I will also use h2testw, maybe I will find the SD card as being faulty.

In gparted it shows the following: https://imgur.com/a/GXCxzCr as if the SD card is 'healthy'

The SD card is mounted on /dev/sdc but in Etcher is non usable and non selectable.

 I will try the last alternative if all the other solutions don't work.

Thank you for all the logical 'approaches' for my problem :)

lurch, 2019-01-09:
> The SD card is mounted on /dev/sdc but in Etcher is non usable and non selectable.

That's definitely a bug, please create a new issue at https://github.com/balena-io/etcher/issues

lurch, 2019-01-09:
...and I think if you had an MBR, gparted would show the "Partition" as `/dev/sdc1` (and not `/dev/sdc`). But that shouldn't affect Etcher in any way.

GabrielBalan90, 2019-01-09:
@Lurch & @procount 
You we're both right. Linux didn't do a MBR on the SD so I used Windows SD Formatter and it did the trick. 
@Lurch you we're right that in Etcher it should show as /dev/sdc1, it seems to be the case after you use Windows SD Formatter the second time. The image problem in Etcher still persists, it's not even showing /dev/sdc1 just /dev/sdc. The issue was raised today and I will be compliant with the explanations and requests.

procount, 2019-01-15:
Can this issue be closed now?

GabrielBalan90, 2019-01-15:
> Can this issue be closed now?

Yes. Please close. Thank you for your help guys 🥇

closed 2019-01-15 by procount