procount/pinn #218: Some strings aren't properly translatable

lurch, 2019-01-08:
While trying to search for relevant info while looking at #217 I noticed you're doing things like:
```
ui->plainTextEdit->appendPlainText(tr("Cannot find setup script for ")+os_name);
```
and
```
ui->plainTextEdit->appendPlainText(os_name+tr(" setup script needs to be updated."));
```

When it comes to translatable strings, that's basically a big no-no :wink: (Once upon a time I worked on translation-related software...)

Just because `os_name` happens to be at the front of one string and at the end of another string in English, it might need to be in a totally different place in another language. That's where the `%1` used in some of the other translatable strings comes in - it acts as a placeholder so the translator can choose where in the string `os_name` (for example) needs to be, rather than forcing it to be at the beginning or end.

I also noticed that there are some strings that don't even use the `tr` stuff at all (e.g. https://github.com/procount/pinn/blob/master/recovery/main.cpp#L319 ), but I won't bother creating a separate issue for that.

procount, 2019-01-08:
Yes, you're right, I hadn't taken that into account as my translations are quite basic.
I'll have to trawl through and check them. Thanks for the tip.

procount, 2019-01-08:
That last comment is due to copying the same code from NOOBS:
https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L246

lurch, 2019-01-09:
Nice catch. Have just created https://github.com/raspberrypi/noobs/issues/527