procount/pinn #219: Migrate from Berryboot to PINN

XxAcielxX, 2019-01-09:
Hello,

I'm a long time Berryboot user, but now I move to PINN.

List of OS I have in Berryboot:
1. LibreELEC
2. Arch Linux ARM
3. Raspbian Lite running rtorrent+rutorrent as SeedBox when I need like.

I know I can start from scratch, but it's time consuming to do all the settings for each OS. Is there a way I can move my files from Berryboot to PINN?

Regards~

procount, 2019-01-10:
I would suggest starting with a new SD card. Install PINN on it, then use PINN to install fresh copies of the 3 OSes that you have installed on Berryboot. 
Now you **_just_** need to copy the files you need for each OS over from Berryboot to PINN 😄.

Now, since Berryboot uses an aufs overlay type file system, you may find it best to boot your Pi using Berry boot and selecting one of your OSes. Then insert you new PINN SD card into a USB slot using a USB card reader and mount the respective partition for the matching OS. Then copy the files you need.

If you have installed any applications, you may want to install fresh versions of them in your PINN version, then copy over any additional data or configuration files you want from Berryboot.

Depending on what you have on your OSes, you may be able to copy the whole root partition over and completely replace the newly installed OS on PINN.
Bear in mind that partitions or mount points might be different between the 2 OS versions and may need to be adjusted. ymmv.

This way you can experiment with your new SD card until you get it right, without wrecking anything on your Berryboot card.

lurch, 2019-01-10:
Also bear in mind that if Berryboot and PINN use different versions of the OSes, the config files for some programs _might_ have different names or be in different places? (and in extremely unlucky cases the config files may not even be compatible if the versions of the applications are different enough :man_shrugging: )

And in addition to what @procount suggests above (I agree with his suggestion of using separate SD cards), another way to do things would be to boot up your Berryboot SD card, copy off whatever files / data you need onto a separate USB stick, then boot up your PINN SD card and copy the data off USB stick onto the new card (which might make the knowing-which-partition-is-which a bit easier?)

scriptxotic, 2019-01-20:
If you want to add new OSes and keep the old ones, you can also create a boot chain : pinn > berryboot.
Simply follow instructions [here](https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version).
You can use `pinn-lite/os/ProjectSpace/partitions.json` as a template file.

closed 2020-04-02 by procount