procount/pinn #220: Autoupdate Auto Update Option?

MartinX3, 2019-01-16:
Dear developers,

I'm sorry, if I didn't found this function, but is it possible to let PINN automatically update itself, if it finds an update?
I don't want everytime to grab my remote and point to the accept button, if there is a new update.
It should just immediately update.

procount, 2019-01-16:
No. Not at the moment.
There is an option for it to NOT check for updates but it is necessary to accept the update manually.
But it looks like a nice feature so I will look to add it for those who like to live at the bleeding edge!

MartinX3, 2019-01-16:
Aren't there different update channels to avoid bleeding edge?
Which versions should I avoid?
I just install every update which comes. 
Except the "can't update itself anymore" bug, it just worked at my site.
(It's only starting libreelec and raspian is installed, but not used)

procount, 2019-01-16:
> Except the "can't update itself anymore" bug,
Can you remind me what this is please?

MartinX3, 2019-01-16:
@procount you fixed it last year or earlier.
We needed to install the new update by hand, because the updater did not mention any update anymore.

procount, 2019-01-16:
Oh ok. If I have fixed it that's fine. i thought it was still outstanding :wink:

lurch, 2019-01-16:
> I think I may call it "forceupdate"

Perhaps "forcepinnupdate" (or similar) to make it clear that it only force-updates PINN and doesn't update any of the OSes?

procount, 2019-01-16:
Forceupdatepinn ?
Iirc the option to update the OSes is "silentinstallnewer" so it is already quite different, but I suppose in isolation it's maybe not so clear.

MartinX3, 2019-01-16:
@procount
Won't upgrade libreelec or raspian itself?
Will I need to reinstall them? Data loss?

procount, 2019-01-16:
@MartinX3 - please stay away from `silentreinstallnewer` :smile:

Once you have installed LibreElec or Raspbian, you can upgrade them using the standard apt-get type tools. You do not need PINN to update them. PINN only restores to a factory fresh version of the OS, (or restores a backup if you have made one).

`silentreinstallnewer` WILL cause data loss, because it will replace the OSes named in the `select` option with the latest factory fresh version of that OS, thus wiping out any modifications you have made or data you have stored indiscriminantly!

I hope I have put enough warnings about this in its description in the documentation:
https://github.com/procount/pinn/blob/master/README_PINN.md#auto-reinstalling-oses

It is for a very specific use case where the OS needs to be updated regularly and it doesn't contain any data. In this case the data would normally be stored in a separate data partition. It doesn't apply to any of the OSes that PINN provides. It would be for a more dedicated OS that is able to automatically find and mount its data partition. Of course there are other ways to do such a thing (e.g. docker or resin-io {balena now?}) but this is a very simplistic way to do it with PINN.

lurch, 2019-01-17:
> (e.g. docker or resin-io {balena now?})

Yeah, I believe resin-io is now balenaCloud.

procount, 2019-02-27:
forceupdatepinn is now implemented in p3.0.4.

closed 2019-02-27 by procount