procount/pinn #222: fatCP437: Invalid argument

YenLegion, 2019-01-18:
I am getting this line when performing a filesystem check upon first installing PINN. I am using a brand new so card. I ran a full format in the official SD format application. I wiped the card once with a quick format then ran chkdsk but still am having the same error. What might be causing this? Here is an image of the disk check results. 

https://imgur.com/gallery/XZcUE16

Thank you.

procount, 2019-01-18:
I haven't seen that error before and I only found 1 google entry about it which wasn't very helpful.
It seems to happen before the file check. Maybe it is to do with code pages?
What size SD card are you using?
Have you changed the language and/or keyboard layout used in PINN?
In any case, it doesn't seem to cause a problem because the fsck completes without errors. So I wouldn't worry too much.
I will look into the code to see it I can spot anything but it might be difficult to debug if I cannot replicate it.

procount, 2019-03-15:
No response - closing

closed 2019-03-15 by procount