procount/pinn #223: pinn 3.0.3 - can not backup usb drive not show maintence?

lpt2007, 2019-01-18:
Hi,

I try to backup to usb drive, I try 64Gb and 16Gb formated to fat32. I cen not select destination drive (nothing to select). Why?
[image]

procount, 2019-01-18:
Did you create a folder on your USB drive called /os?

lpt2007, 2019-01-18:
Thanks I add os folder on usb and now I see destination drive.

Now I have another error. What this error menas?
[image]

procount, 2019-01-18:
Ah yes. You picked one of the few OSes I can't back up yet.
Xbian uses a btrfs file system which requires some very special code to back it up. I hope to have that sorted soon but it is a bit tricky.

lpt2007, 2019-01-18:
Ok, Thanks.

+1 vote for xbian backup solution

closed 2019-01-18 by lpt2007