procount/pinn #227: Raspbian disk space

prodigy213, 2019-02-10:
Hello,

I would like to know why Raspbian is creating a 2.9GB root directory on the SD card. The NOOBS support page says there should be a minumum 4GB for Raspbian. During the OS install there was no red text indicating a space issue. At the moment I cannot update/upgrade Raspbian due to lack of disk space. Before displaying my disk space, is there a way to tell or adjust PINN tell it how much disk space each OS can take up during install instead of having it automated? For example have Raspian take up 4GB Raspbian Lite 512MB, Retropie 2GB, etc.

I got the out of disc space message while running this command in terminal:

sudo apt-get update -y && sudo apt-get autoclean -y && sudo apt-get upgrade -y && sudo apt-get autoremove -y

Thank you.

Below is the output of df -h.

Filesystem       Size  Used Avail Use% Mounted on
/dev/root        2.9G  2.7G     0 100% /
devtmpfs         426M     0  426M   0% /dev
tmpfs            430M   51M  379M  12% /dev/shm
tmpfs            430M  6.0M  424M   2% /run
tmpfs            5.0M  4.0K  5.0M   1% /run/lock
tmpfs            430M     0  430M   0% /sys/fs/cgroup
/dev/mmcblk0p19   68M   23M   46M  33% /boot
tmpfs             86M     0   86M   0% /run/user/1000
/dev/sda1         60G  4.4G   56G   8% /media/pi/AE90-D3C3
/dev/mmcblk0p9   4.7G  4.5G     0 100% /media/pi/root_pitop
/dev/mmcblk0p8    43M   23M   21M  52% /media/pi/boot_pitop
/dev/mmcblk0p7   4.0G  3.7G   94M  98% /media/pi/root_cstem
/dev/mmcblk0p6    42M   21M   21M  51% /media/pi/boot_cstem
/dev/mmcblk0p5    30M   17M   11M  61% /media/pi/SETTINGS
/dev/mmcblk0p31  488M  408K  452M   1% /media/pi/data
/dev/mmcblk0p30  446M   13M  406M   3% /media/pi/root_lede3
/dev/mmcblk0p29   20M  9.4M   11M  47% /media/pi/boot_lede3
/dev/mmcblk0p28  1.5G  1.1G  239M  83% /media/pi/root3
/dev/mmcblk0p27   68M   23M   46M  33% /media/pi/boot3
/dev/mmcblk0p26  2.2G  1.3G  751M  64% /media/pi/root-rbp2
/dev/mmcblk0p25   79M   24M   56M  30% /media/pi/boot-rbp2
/dev/mmcblk0p24  2.1G  717M  1.3G  36% /media/pi/pmb_root
/dev/mmcblk0p23   60M   22M   39M  37% /media/pi/pmb_boot
/dev/mmcblk0p22  701M  353M  298M  55% /media/pi/Storage0
/dev/mmcblk0p21  512M  143M  370M  28% /media/pi/System0
/dev/mmcblk0p18  2.1G  1.4G  561M  72% /media/pi/root1
/dev/mmcblk0p17   70M   22M   49M  32% /media/pi/boot1
/dev/mmcblk0p16  1.7G  1.3G  263M  84% /media/pi/root0
/dev/mmcblk0p15   70M   22M   49M  31% /media/pi/boot0
/dev/mmcblk0p14  206M  124M   68M  65% /media/pi/share
/dev/mmcblk0p13  1.9G  1.2G  681M  63% /media/pi/root
/dev/mmcblk0p12   60M  7.1M   53M  12% /media/pi/boot
/dev/mmcblk0p11  701M  325M  326M  50% /media/pi/Storage
/dev/mmcblk0p10  512M  298M  215M  59% /media/pi/System

lurch, 2019-02-10:
> The NOOBS support page says there should be a minumum 4GB for Raspbian.

That might be out of date, I think these days 8GB is recommended for Raspbian? @XECDesign ?
https://www.raspberrypi.org/documentation/installation/sd-cards.md

procount, 2019-02-10:
Wow, that's a lot of OSes you have installed there!
I set the minimum size of the rootfs partitions to have about 500MB free, which is normally just enough to get it to run. But if you're going to install lots of apps or update etc. then you need to ensure you have enough disk space. Any remaining space on the sd card is divided equally amongst the OSes, so the less you install, the more disk space each will have.
If you want to distribute the space differently, then I recommend you visit pinn.mjh.net to select your OSes. Matt's Web App allows you to customise the space allocated.

XECDesign, 2019-02-11:
Raspbian got the 'full' version to reign in some of the space used up by the large applications. The regular version should fit on 4GB, but I don't think it leaves enough wiggle room. 8GB is recommended as a comfortable minimum.

lurch, 2019-02-11:
> The NOOBS support page says there should be a minumum 4GB for Raspbian.

@prodigy213 Can you point us to where you saw that please? I just had a good hunt around, and everything I can find says "8GB or larger". The only place I was able to find a reference to 4GB was https://www.raspberrypi.org/documentation/installation/sd-cards.md which says "For Raspbian Lite image installations we recommend a minimum of 4GB."

prodigy213, 2019-02-11:
> @prodigy213 Can you point us to where you saw that please? I just had a good hunt around, and everything I can find says "8GB or larger". The only place I was able to find a reference to 4GB was [...]

@lurch Sorry for the confusion. I was typing from memory I wasn't sure if it was Lite or Desktop that required 4GB. I was skipping support pages a while back about disk space too so thats where I had that idea. I would have doubled or tripe the amount of disk space because I tend to load a lot of thing.

prodigy213, 2019-02-11:
> Wow, that's a lot of OSes you have installed there!
> I set the minimum size of the rootfs partitions to have about 500MB free, which is normally just enough to get it to run. But if you're going to [...]

@procount Thank you. I consider this reply satisfactory and will close this issue. I have a 64GB and just purchased a 128GB for the other Pi I have. Indeed the the installation was proper on the 64GB. 

I just have one last comment/question, I assumed everything would be okay because there were no red flags or warning during the installation (When I selected too many OS there were red text and it did give me a message during the OS installation and also if there wasn't enough disk space to complete the installation). It would be a nice feature to have an option to ensure Raspbian Desktop and other "big" OSes (like C-STEM) get the minimum required space they need so this won't happen to other rookies like myself. (But I love the current option too because I got to cram a lot onto a 32GB SD card so I know I will regret it. :1st_place_medal: ). I just saying this because I assume people in financial strain will attempt to do what I just did.

procount, 2019-02-12:
Yes, I check you don't exceed the highest bootable partition number.

It's difficult to define an absolute minimum partition size for each OS. Whilst 4GB or 8GB might be _RECOMMENDED_ for Raspbian, you can get away with less depending on what you want to do. If I set the minimum to 8GB, would a newbie expect it to work on an 8GB SD card? I would actually have to set the min size to less than 8GB to account for the 64MB taken up by PINN, and a little more to account for the variation in 8GB card sizes. And suppose they wish to reserve a little space on PINN's partition for the installation of a small OS's installation files, like void - should I anticipate that as well?

There are so many variations, that the minimum size is set to have about 500MB free to allow the most number of OSes to be installed. (Actually, this is the same value that is set within NOOBS for Raspbian as I just share those same files). But it does put the onus on the user to ensure they have allocated enough space for what they want to do. I heartily recommend Matt's webapp ([http://pinn.mjh.nz](http://pinn.mjh.nz)) to distribute the remaining free SD card space amongst the installed OSes if you need complete control over the storage allocation.

@XECDesign, @Lurch, - One idea that comes to mind is adding an optional "Recommended_size" option in partitions.json/os_list_v3.json. Then users can be warned if an OS is being allocated too little space, but they would not be prevented from doing so (as that is already catered for in the nominal_size). Optionally, partition allocation could be done on recommended_size instead of nominal_size. Of course, the OS maintainers would have to determine and add this value. What do you think?

lurch, 2019-02-12:
> Of course, the OS maintainers would have to determine and add this value. What do you think?

I think most of the OS maintainers are lazy (at least WRT NOOBS), and they'd set this value once but never update it for subsequent OS releases, which would make it pretty useless. When I was the maintainer for NOOBS (which was a while ago now!), OS maintainers would send me updated tarballs without updating any of the metadata files, so I'd have to update all the metadata files by hand :slightly_frowning_face: 

And of course there's no way to predict what a user is going to do or install onto their OS, so no matter what you set the recommended size to you'd likely still have some people complaining it was too small :wink:

procount, 2019-02-12:
Yes, you can never please all the people, all the time!
So, another option is to incorporate a cut-down version of Matt's webapp, to visualise and allow adjustment of the allocated partition sizes ala gparted / partition magic etc. (but maybe not so fancy)
And/Or maybe set an artificial recommended_size of 2-3x nominal size?

prodigy213, 2019-02-12:
@lurch @procount @XECDesign Thanks for helpfing and clearing things up for me and providing me with suggestions and options.

> Yes, you can never please all the people, all the time!

@procount I don't want it to be too much of a hassle. I love to being able to over load the SD cards. I hope this ticket will help someone else who "loaded up" their SD cards and can't update their OSes due to disk space. This should provide them with options and solutions. (1) Assume each OS you load will take between 4-8GB (unless specified) plus 64MB taken up by PINN (2) if they have free space they can use Matt's webapp (http://pinn.mjh.nz) (3) find a bigger SD card to load more OSes onto.

> So, another option is to incorporate a cut-down version of Matt's webapp, to visualise and allow adjustment of the allocated partition sizes ala gparted / partition magic etc. (but maybe not so fancy)
> And/Or maybe set an artificial recommended_size of 2-3x nominal size?

That would be lovely!!! "Advanced users" can select the amount of space needed and amateurs can ignore and have it automated.

lurch, 2019-02-13:
Just to clarify - I believe "Matt's webapp" isn't for "if they have free space", but for "if they want to redistribute the free space between OSes differently to how PINN does it by default" ? :wink:

procount, 2019-02-13:
Absolutely.
There is never any free space after a noobs or pinn install. All the sd card is allocated. It's just a matter of distribution

closed 2023-03-26 by procount