procount/pinn #228: PgDown crashes PINN

dominic-03, 2019-02-11:
Yesterday when I was in the PINN menu of operating systems, I pressed PgDown on my keyboard by accident. It cut the display output to my TV and my Pi stopped responding. After trying it again, I can confirm it can be replicated. Pressing PgDown again after the display loses signal does not turn it back on.

This is a odd bug, so I just wanted to write it on here.

procount, 2019-02-11:
Which model of RPi do you have?
What version of PINN are you using?
Were you in the "Main Menu" menu of PINN (with the tabbed dialog box of categories of OSes, rather than the Boot menu)?

dominic-03, 2019-02-11:
> Which model of RPi do you have?
> What version of PINN are you using?
> Were you in the "Main Menu" menu of PINN (with the tabbed dialog box of categories of OSes, rather than the Boot menu)?

I edited my post with that information now included.

procount, 2019-02-11:
I just tried it on mine and it didn't crash, so something funny is going on with yours.
Please have a look on my troubleshooting section to see if you can capture any logs of interest.
https://github.com/procount/pinn/wiki/Troubleshooting

But I see PageDown is doing 2 things: 1) moving the highlight to the bottom of the current list and 2) activating the Next Menu button, so maybe I will remove that functionality in the next version and rely on the 'M' key alone.

lurch, 2019-02-11:
@dominic-03 What keyboard are you using? What language do you have PINN set to?

dominic-03, 2019-02-11:
> @dominic-03 What keyboard are you using? What language do you have PINN set to? Is your Pi connected to your TV via HDMI or composite?

I'm using HDMI TV
I have PINN set to United Kingdom English and I have a standard Dell US keyboard. Do I need to set the language to US English?

> I just tried it on mine and it didn't crash, so something funny is going on with yours.
> Please have a look on my troubleshooting section to see if you can capture any logs of interest. [...]

I'll try it again tonight, and get more details on it.

lurch, 2019-02-11:
@dominic-03 That all sounds fine, I just wondered if maybe you were using some obscure keyboard which was sending a non-standard keycode which PINN wasn't expecting. (I think quite a few people use PINN for a media center and use a wireless keyboard, so I was just checking)

dominic-03, 2019-02-12:
Also, if PINN stands for "PINN Is Not Noobs" then what does PINN stand for in the acronym? (You may be confused, and yeah, this does sound pretty confusing as I type this."

lurch, 2019-02-12:
It's a recursive acronym - pretty common in the computer world :wink: 
https://en.wikipedia.org/wiki/Recursive_acronym :nerd_face:

dominic-03, 2019-02-12:
> It's a recursive acronym - pretty common in the computer world 😉
> https://en.wikipedia.org/wiki/Recursive_acronym 🤓

Oh, ok. Cool. (You learn something every day!)
I added PINN to that article, just for the fun of it.

procount, 2019-02-12:
Credits to Lurch for suggesting the name in the fiest place. :wink:

procount, 2019-03-15:
v3.1.1: Using pagedown to cycle the menus has now been removed, as pagedown is also used to move down the list. So it is necessary to use the 'm' key to cycle the menus now.

closed 2019-03-15 by procount