procount/pinn #229: Black screen after Combian 64 boots

mobluse, 2019-02-18:
Unfortunately I get a blank/black screen with background LEDs off, but the screen doesn't go into sleep mode (the LED is still blue), when Combian 64 boots, and then it stays black. Amibian and other games OSs work, but some also have a black screen. I can use PINN if I press Shift at boot or if I multiboot. I use a Raspberry Pi 2. I have a screen with HDMI to DVI-D adapter. The screen is 1280x1024, 64 kHz / 60 Hz.

lurch, 2019-02-18:
Just out of curiosity, have you tried the image-version of Combian 64 (never used it myself) on the same hardware? Does that boot okay?

mobluse, 2019-02-19:
No, the Combian 64 image version also gives a black screen with most monitors I have (60 Hz and above), and the PINN and the image version works with one monitor I have (50 Hz and above).  It seems the distros that doesn't work have a slightly lower vertical frequency e.g. 59 Hz. This is not a problem with PINN, but with some distros. E.g. Amibian works and that also has a VICE C64 emulator, if you install the bonus pack. About half of the games distros doesn't work with my 60 Hz screens.

The monitor Combian 64 works with is HP Compaq LA1951g.

I never had these problems with black screens before, but I have not used those distros that are problematic now before. I think all PINN distros should work with my normal 60 Hz screens that are using HDMI to DVI-D adapters.

lurch, 2019-02-19:
Well if the image version and the PINN version **do** work on at least one of the monitors you have, then it sounds like this is a Combian 64 problem rather than a PINN problem? :wink: 

I guess you might want to fiddle with config.txt and hopefully you'll find some settings that work for you?
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

procount, 2019-02-20:
Yeah, sounds like a Combian 64 (and other emulator OSes) issue, especially as the same behaviour is present  whether it is installed by PINN or directly as an image.
In fact I would expect the same behaviour, because all PINN does is to repackage the OS image into a different format. The OS is not altered at all. When PINN does make changes, it is only to allow older OSes to boot on older RPi models where the latest firmware does not work, or to remove UBoot in order for it to boot from its new partition location.

So I shall close this issue. Please feel free to re-open it if you find a new issue with PINN.

closed 2019-02-20 by procount