procount/pinn #230: Auto-Reinstalling OSes

Fateofiko, 2019-02-18:
Hi !
Sorry, if i am missing something, but i tried to "silent reinstall" the only Os, that i have installed on the SD card, and every time after the installation, the menu which displays all installed OSes shows up and do not boot the OS.
Is there a way that i can reinstall the OS and automatically boot it?
I am testing on my raspberry, which have a display, but i want to do this on my headless raspberry!
I set this options to recovery.cmdline :  
quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json no_update reserve=+2000 disableusbimages no_default_source bootmenutimeout=10 select="Raspbian Lite" silentreinstallnewer forcetrigger
I tried and other options, but wasn't able to boot the OS automatically after reinstall!

lurch, 2019-02-18:
Sounds like you want #221 ?

procount, 2019-02-19:
Yeah, I don't think reinstalling (silent or otherwise) goes to the boot menu. It may need another option. I'll take a look when I can but it might have to wait a week.

procount, 2019-02-19:
Forcetrigger is not helping. Did you try remotemenutimeout (or something like that)? But it still may not work even with rhat.

lurch, 2019-02-19:
@procount I was assuming that he always sees the menu because he has `forcetrigger` in there?
(although I've never played with PINN, so apologies if I'm misunderstanding and simply spreading more confusion! :confused: )

procount, 2019-02-19:
Could be. On a plane in HK so can't easily check atm. ;)

Fateofiko, 2019-02-19:
@lurch Yes, i saw #221, but i do not want to reinstall on every reboot! And about "`forcetrigger`" if i do not use forcetrigger the PINN just waits for me to interact with shift key, otherwise it just continue to boot the existing OS.
The option "runinstaller" with "silentinstall" works perfect, but i thought i can skip the format of the whole SD card, but i do not care because after the installation it boots the OS automatically.
Is there a reason, that i should not use the "runinstaller", if i do not care of the re partitioning the SD card?
@procount Thanks, i will try and with "remotemenutimeout" :+1: ! And do not worry i can wait, i am just happy for your great support of the project :) !

lurch, 2019-02-19:
@Fateofiko I think part of the problem (AIUI!) is that (as the name implies) `silentreinstallnewer` will only reinstall the OS if the version available (e.g. from the internet or the SD card) is newer than the version you already have installed. And since the version you have installed is the same as the version you're trying to reinstall, PINN ignores it and "it just continue to boot the existing OS".

@procount Is there some json file (e.g. on the SETTINGS partition) that @Fateofiko could edit to 'trick' PINN into thinking that the installed version is older than the available version? If I've understood what he's asking for, I think that might be a workaround.

Fateofiko, 2019-02-19:
@lurch @procount  yes there is  /settings/os/raspbian_lite/os.json file with release_date, and i am changing it every time i do reinstall, but without "forcetrigger" it waits me to interact with the shift key!
If i set "forcetrigger" but do not change the release_date the reinstalling do not start, which is logical :) !
About "remotetimeout=10" with "forcetrigger" i tested that, and 10 seconds after the progress bar for re installation process starts, the whole graphic interface hides, and after an hour of waiting there is no auto booting of OS! 
I think i am going to use the "runinstaller" with "silentinstall" it works fine for my purposes!
But thanks again, i appreciate your help :) !

lurch, 2019-02-19:
> About "remotetimeout=10" with "forcetrigger" i tested that, and 10 seconds after the progress bar for re installation process starts, the whole graphic interface hides

@procount Does that mean that the `remotetimeout` is cancelling the installation process? If so, that sounds like a bug :wink:

Fateofiko, 2019-02-19:
@lurch I am not sure that it cancel the installation process, it just hides the graphic interface and shows the terminal log. I waited more than hour, and then restart the unit manually!
After the restart everything happen again, and it stuks to the terminal without to boot any OS. After that i had to put the SD card to my laptop to change the recovery.cmdline file, and after that it boot the OS normally!
The only thing that i can say is, this combination "remotetimeout=10" with "forcetrigger" and "silentreinstallnewer" and " select="XXXXXX" " do not work, but you can make your own test, may be i am doing something wrong :) !

procount, 2019-02-19:
I think of it more as an unexpected use case. ;)

procount, 2019-02-20:
OK, so looking at the code only, the silentreinstallnewer will not cause a reboot after reinstall, because it re-uses the reinstall routine, which also doesn't reboot at the end.
I'm not sure that I want to force a reboot after a manual reinstall of some OSes, so I'll have to make a distinction between the 2 methods of reinstalling.
Should silentreinstallnewer ALWAYS cause a reboot, or should I make it another option?

lurch, 2019-02-20:
When you say "reboot" do you mean an actual reboot, or "continue to the OS boot selection screen and boot the default OS"?
IMHO if you're using any "silent..." option, then it makes sense to continue and boot the default OS, because I guess "silent" possibly also implies "headless" ?

procount, 2019-02-20:
See #156 for the original discussion.
Since all the detection code is in mainwindow.cpp, forcetrigger is required to enter the menu to check for new OSes. Remotetimeout is then also needed so that under normal circumstances, the default OS will still boot. So normal startup with no update is a bit delayed.
(It would be nice to change this so the menu would not need to be entered, but I think it is too big an architectural change. Especially since the original requester never replied and this is probably the first use of this feature since then!)
I think what I missed was the need to reboot after reinstallation. (If I go to the boot select dialog from the recovery menu, the timeout timer will not normally work, so it will just sit there waiting for user input. I could make a special case to force the timer to boot the default OS, or I could just reboot PINN instead, which will take a bit longer, but require less modification, or I could grab the partition number of the default OS and boot straight into it)
And also I the need to prevent multiple OSes installed into ProjectSpaces from all being updated if that is not required.

lurch, 2019-02-21:
A perfect example of how complex things get when you start adding more features, especially when they interact in unexpected ways :wink: :rofl: 
I myself hadn't even considered the implications of `silentreinstallnewer` when more than one OS is installed... (which of course complicates the decision about which OS to boot afterwards).

Having just quickly skimmed through #156 it sounds like "If exists os with overwrite config, execute its installation, including overwrite of partitions as listed in its config file, and boot into the new OS" is also exactly what @Fateofiko is asking for?

Good luck with getting it all sorted out! :+1: :crossed_fingers:

procount, 2019-02-21:
I'll need several strong coffees before attempting this :smile:

lurch, 2019-02-21:
...and make sure you've recovered from your jetlag too! :grinning:

Fateofiko, 2019-02-27:
Simply, i just need after reinstalling the OS, just automatically to boot in the default one. In this case, i am using "runinstaller", this option is perfect!
Why i am using this, the reason is simple:
My headless raspberry, pings on every N seconds to remote server, and if i need the raspberry to do something,  i am setting up the server to response with commands in json format.
One of the commands starts a script, which downloads a new tarballs with the configuration files and replace everything in the os folder in the recovery partition of PINN, and also replace the recovery.cmdline file with new one, with other configuration.
The script automatically restarts the raspberry, and i need PINN to do his job (install the new tarballs, and boot the OS). After that the new OS automatically starts my custom application, which connects to my remote server.
This way i can recover/reinstall/update the image of the whole system :)!
So the "runinstall" option with "silentinstall" works perfect, and if you think that it is more useful to update the other options like ("silentreinstallnewer"), then you can do it. 
But everyone that needs a solution of this sort, simply can use the "runinstall" option with "silentinstall", of course and the option "reserve" is a big advantage :) !

And again, great project and great support, thank you all!

procount, 2019-02-27:
OK, well it sounds like you already have a solution that suits your purposes, so I will drop the priority on this issue, but I'll keep it open and try and address it later.

lurch, 2019-02-27:
Sounds like you're trying to come up with a DIY version of what https://www.balena.io/ does? :wink: 
(remote OS & application updates)

Fateofiko, 2019-02-27:
@lurch "DIY version of what https://www.balena.io/" , i am not sure what "balena" does, but yes  (remote OS & application updates for raspberry :) ) !!!

procount, 2019-03-15:
From v3.1.1, a silentreinstallnewer will now cancel the remotetimeout timer to avoid it cancelling the installation. Afterwards, PINN will go to the boot menu and start the boot timeout. Provided no user interaction stops it, it will then boot the last OS, or the sticky default, however you have configured it.
So I think this will now cover your use case.
Closing as complete, but feel free to reopen if there is a problem.

closed 2019-03-15 by procount