procount/pinn #231: Replace ProjectSpace now working

Rippert, 2019-02-20:
I have tried to install one OS (ARCH2) and 4 Project Spaces on a USB hard drive, and then replace one of the Project Spaces with Raspbian Lite. The Arch install works and will boot, and the replace finishes saying "reinstall completed". 

I cannot see Raspbian Lite in the list of installs or boot to it. Also, I would expect that the dialog box should say "repalce completed" not "reinstall, so maybe some kind of mixed up command bug?

I'm using a Rpi 3B+

Thanks,
-Ted

Rippert, 2019-02-20:
Just checked inside Arch, and the next two partitions do have all the files for Raspbian Lite in them, they just aren't recognized as an installed system by PINN. It thinks they are still a Project Space.

I looked in the PINN recovery partition too, but I can't find what file I need to edit to tell PINN that partitions 8 and 9 are Raspbian Lite.

lurch, 2019-02-20:
I've not used ProjectSpaces myself, but it sounds like you'd need to edit some files on the SETTINGS partition to get PINN to recognise the extra OS? (although what you describe above definitely sounds like a bug rather than expected behaviour)

procount, 2019-02-20:
Are you using the latest version of PINN?
Did you try rebooting?
I'll check this when I can.

Reinstall and Replace are virtually the same function -it's just a matter of what you are putting in its place - a new copy of the same OS, or a different one. I think I was just being lazy sharing the same message. Sorry for the confusion.

(@lurch - PINN will have updated the appropriate files on /settings. Earlier versions required a reboot for PINN to re-read them all, but latest version shoudn't require a reboot - unless I broke something)

Rippert, 2019-02-20:
I have rebooted repeatedly, no change. 

PINN says v3.0.3 - Built Dec 27 2018

Seems to be ahead of what you have listed as releases on Github. I got it from sourceforge, it's the Etcher version.

I'll try looking at the Settings partition and see what's there.

(Sorry about the typos, GitHub won't let me click the edit button now, probably doesn't like my old version of Safari)

Rippert, 2019-02-20:
OK, looked at settings partition, and /settings/os/Raspbian_Lite directory was there. However, the second entry in /settings/installed_os.json was for a Project Space. So I edited the second entry to match what was in the Raspbian_Lite folder, and it worked fine on the next boot.

So the installed_os.json file apparently doesn't get updated by the "Replace" procedure.

I've also noted that the language chooser at the bottom of the PINN screen keeps reverting to UK english, and I have to keep setting it to US english. Not sure if that's related or just my misunderstanding whether that was a setting that was saved.

procount, 2019-02-21:
Thanks fir investigating. Not sure hiw that happened. Will fix shortly.

procount, 2019-02-21:
I think the language problem is specific to English US. I tried another language and it saved ok, but English US reverts to English UK, I expect it may happen on NOOBS too. I'll have a look.

procount, 2019-02-21:
Investigating the language issue, there is no specific language translation for English US, so it uses the same English UK text instead. But selecting English US language does select the US keyboard layout and this does seem to be saved. So it seems to be working correctly, even if the language displayed reverting to English UK seems to be incorrect.

lurch, 2019-02-21:
Is https://github.com/raspberrypi/noobs/issues/179 relevant? I'm afraid that was so long ago that I can't remember what the resolution was (if any).

procount, 2019-02-21:
Yes that looks like exactly the same language problem. Quick Solution; `cp translation_gb.qm translation_us.qm` (or whatever the filenames are called - you get the idea) :wink:

I'll look at the main projectspace issue when I get back.

lurch, 2019-02-21:
IIRC (and I'm really rummaging around in my memory here - been a long long time since I've actually looked at the code) I think that the issue with languages might be that NOOBS conflates both the 'language' (as used for translating GUI strings) and 'keyboard' (as used for USB keyboard input) settings into a single "Language" GUI option? :man_shrugging: :gb: :us:

procount, 2019-02-21:
Exactly.
I think the idea must have been that if you change the language, then you must also want to change the keyboard layout. But the keyboard can then be changed independently of the language setting. So the EnglishUS language entry tries to change both. Fails to find the US language so it reverts to the uk being the closest, but the keyboard stays as US.

lurch, 2019-02-21:
Word of warning - when playing with this you also need to be careful that you choose 'gb' instead of 'uk' because the latter is the https://en.wikipedia.org/wiki/List_of_ISO_639-1_codes for Ukranian :wink:

procount, 2019-02-25:
> I have tried to install one OS (ARCH2) and 4 Project Spaces on a USB hard drive, and then replace one of the Project Spaces with Raspbian Lite.

@Rippert - Did you do this with PINN installed on the USB hard drive, or was it installed on the SD card?

procount, 2019-02-25:
I've done a couple of quick tests as follows, so not exactly the same, but pretty close:

**Test 1:** 
PINN installed on SD card, 
Installed Arch2 + 4 ProjectSpaces to SD card.
Rebooted & Arch loaded ok.
Rebooted into PINN.
Selected Raspbian (Full) and replaced ProjectSpace 1 with it.

**Test 2:** 
PINN installed on SD card, 
Inserted 16GB USB stick
Installed Arch2 + 2 ProjectSpaces to USB stick (not enough space for all 4)
Rebooted & Arch loaded ok.
Rebooted into PINN.
Selected Raspbian (Full) and replaced ProjectSpace 1 with it.

In both cases, the results were the same:
Raspbian was installed ok.
The maintenance menu did not update to show ProjectSpace1 had been replaced with Raspbian.
Selecting EXIT to go to the boot menu showed the 2 installed OSes correctly.
Raspbian could be booted.
On subsequent boots into PINN, the maintenance menu displayed correctly.

So, I have to try and fix the update of the Maintenance Menu.
The above tests were done with a Pi2 and Raspbian, so they need to be repeated with a Pi3B+ and Raspbian Lite. Also I need to know if PINN was installed on the SD card or the USB device.

Rippert, 2019-02-25:
PINN was installed on a USB hard drive. There was no SD card inserted in the Pi at all. It is a 64GB solid state drive, and I used Etcher on a Macintosh to install PINN to it.

procount, 2019-02-25:
Ok I repeated it. Seems to be related to booting PINN off USB. Will investigate.

procount, 2019-02-26:
Ahh! It's to do with PARTUUIDS. A fix is in progress....

procount, 2019-02-27:
PINN version 3.0.4 now released with this fixed in it.
(I'm going to leave the language issue as it is)

closed 2019-02-27 by procount