procount/pinn #232: Unable to Update Raspbian after installing fresh build...

SalvorinFex, 2019-02-23:
Greetings, 

Having just discovered PINN, I copied the latest build to a micro SD card and setup some distros on a 32gb card.  Raspbian being one of them, I booted into that first, after the initial boot setup, I attempted to run `sudo apt update` and `sudo apt upgrade`, but after the first section of the upgrade, I received this:

```
148 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 103 MB/258 MB of archives.
After this operation, 4,368 kB of additional disk space will be used.
E: You don't have enough free space in /var/cache/apt/archives/.
````

I don't know why it doesn't say it doesn't have enough space, as File Explorer tells me 5gb is free.
Any ideas?

procount, 2019-02-23:
It is very tempting with PINN to install too many OSes. Have a read of #227 where a similar issue has recently been discussed. Feel free to ask further questions here if you need more info.

procount, 2019-02-27:
Closing this as a duplicate of #227

closed 2019-02-27 by procount