procount/pinn #233: Is the Compute Module 3 Supported?

mjjames, 2019-03-04:
Hi,

Great project and lots of additional functionality over default Noobs.

I've treid this using SDCards on a Pi3B but now looking at using it with a Compute Module 3, and soon the 3+. However I can't get it to boot. Is it supported? I've noticed in root of the project there is a dtb for the cm3, is there any specific config required for it to use this? 

I am using the PI touch display and not HDMI could this be the issue?

Thanks

procount, 2019-03-04:
I don't have a Compute Module of any type, so I've never been able to test it definitively. But I see no reason why it shouldn't work since NOOBs apparantly works with them. The dtb is already provided and will be selected automatically by the firmware.

I use the RPF 7" touchscreen display myself, so I know that works with PINN on a standard (i.e. not a CM) RPi model.

Sorry I can't really provide much advice. I suggest you try getting NOOBS to work with your CM first. Once that works, PINN should just work the same.

mjjames, 2019-03-04:
thanks for the quick reply. Exactly the steps I'm now going through, CM3 developer board with HDMI and then all the steps before using our carrier board.

I'll keep this open in case I find anything and can advise further. I'll close once I know the solution or where the problem is.

procount, 2019-03-04:
I would be interested to know of any outcome since I am not aware of anyone using PINN with the CM. But then the developer only gets to hear of problems, rather than successes :wink:

mjjames, 2019-03-04:
As I initially suspected. The issue is down to overlays and device pins. I took our dt-blob.bin and dropped it into the recovery partition the screen loads fine. I originally used https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-display.md to build this dt-blob. 

This should now be possible by just loading an overlay but I'm not sure if there already is one or not. I'll advise further if i find an overlay that could just be included and then enabled via the config.txt file

procount, 2019-03-04:
Ah! Some CM3 trickery is required for the touchscreen. I guess HDMI output would have worked ok.
Out of interest, without the dt-blob or overlay, does /sys/devices/platform/rpi_ft5406 exist when the touchscreen is connected? Or does it only appear with the dt-blob present?

lurch, 2019-03-04:
@pelwell Can you provide any relevant advice here? :wink:

mjjames, 2019-03-04:
Yes, HDMI worked fine its what caused me to revisit the overlays and dt-blobs. I've just had confirmed on the Raspberry Pi forums that you have to provide a dt-blob.bin . In this case is it worth updating the documentation to include a reference to https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-display.md ?

pelwell, 2019-03-04:
@6by9 has answered this on the forums.

lurch, 2019-03-04:
@mjjames Which documentation are you suggesting should be updated?

procount, 2019-03-04:
I think he is referring to the use of the RPF touchscreen in NOOBS/PINN which requires the dt_blob.bin. So therefore the NOOBS / PINN documentation.

mjjames, 2019-03-04:
sorry yes, NOOBS / PINN

procount, 2019-03-04:
And don't forget my question:
> Out of interest, without the dt-blob or overlay, does /sys/devices/platform/rpi_ft5406 exist when the touchscreen is connected? Or does it only appear with the dt-blob present?

lurch, 2019-03-04:
Yeah I'd tend to agree - I think the small number of people who are going to be running PINN on a CM while using a touchscreen is so small, that it's probably not worth updating the PINN documentation to cover it. (and if anyone does have problems, they'll hopefully do a search and find this issue :wink: )

closed 2019-03-15 by procount