procount/pinn #238: recovery.cmdline ignores settings attached with newline

MartinX3, 2019-03-25:
Dear developer,

sadly settings / options / configs attached as a newline in the recovery.cmdline are getting ignored.
I prefer newlines to put the commands in groups or to put every command in its own line.
It's easier to read for me.
Having everything in one line is a bit hard to read for me.

Greetings
MartinX3

procount, 2019-03-25:
Yes, I'm afraid all the options must be included on one single line.
Changing that would touch too much of the code.
But also the number of options now is getting too large to remember, so I have a plan on my roadmap to resolve this, but in a different manner.

MartinX3, 2019-03-25:
Like a GUI in PINN to set the options? <3

lurch, 2019-03-25:
> Yes, I'm afraid all the options must be included on one single line.
Changing that would touch too much of the code.

Isn't `recovery.cmdline` (which is the NOOBS equivalent of `cmdline.txt`) read by the RPi firmware? AFAIK this means @procount couldn't modify the "everything has to be on one line" behaviour even if he wanted to? :wink:

MartinX3, 2019-03-25:
Or just replace every newline with a space if we click on "okay" :P

procount, 2019-04-10:
Fixed in V3.2

closed 2019-04-10 by procount