procount/pinn #239: Problem with ScreenSwitch

Denethor85, 2019-03-27:
Hi,

I extracted the zip file into root of SD.
When i power on the raspberry without the hdmi and start Raspbian switch the resolution of LCD but the signal is in HDMI.

Sorry for my English

procount, 2019-03-27:
Which lcd screen do you have?
You may have to copy one of the config files to config.txt. Try copying config.hdmi to config.txt (do it on your PC for ease).

Denethor85, 2019-03-28:
I have a 3.5 LCD buy from China...

procount, 2019-03-28:
From https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1440283#p1440283

"ScreenSwitch.zip is a package of files that allow PINN to detect and switch between 4 display types: HDMI, DSI (rpf touchscreen) and the Hyperpixel 3.5" and 4.0" screens at bootup."

It's only for those displays.

closed 2019-03-28 by procount