procount/pinn #240: Recalbox install on RPI 3B (not+)

daviduco, 2019-03-27:
Can't install Recalbox, it always says can't download meta.

I've checked installation paths and these files are missing.

http://pinn.mjh.nz/os_list.json?recalboxos-rpi3=50332&osmc_pi2=7168

https://archive.recalbox.com/v1/noobs/partition_setup.sh
https://archive.recalbox.com/v1/noobs/recalboxOS.png
https://archive.recalbox.com/v1/noobs/rpi3/boot.tar.xz
https://archive.recalbox.com/v1/noobs/rpi3/root.tar.xz
https://archive.recalbox.com/v1/noobs/marketing.tar
https://archive.recalbox.com/v1/noobs/rpi3/os.json

procount, 2019-03-28:
It looks like they've removed the NOOBS install version from their website.
Thanks for letting me know. I'll see if I can find out.
Meanwhile, have you tried Batocera, or one of the other game distros?

daviduco, 2019-03-28:
I tried to put recalbox image in os folder but as soon as it connects to internet it tries to get online version. If i remember well on noobs it priorized offline files.

I'll try batocera this afternoon.

Thanks for the quick answer =D

procount, 2019-03-28:
PINN also prioritizes the offline version if the date is the same or newer.
Did you download the os files before they disappeared from the recalbox website?

daviduco, 2019-03-28:
I downloades recalbox files last night and it skips them as soon as i connect to internet.

procount, 2019-03-28:
It just checks the release date.

I guess if you tried to download them last night then they are incomplete, since they are missing on the website.

I have messaged recalbox to find out what has happened to them.

daviduco, 2019-03-28:
I don't get it. If offline version is priorized then PINN should use offline files.
I downloaded os zip from recalbox githubt. Installed it without pinn and it works like a charm.

lurch, 2019-03-28:
> i installed it without pinn

The files needed for standalone image-installation are very different to the files needed for a PINN-installation of the same OS.

procount, 2019-03-28:
OK I remember now!
Recalbox  was removed from NOOBS because it would no longer work with v2.9 due to it using very old firmware, but I managed to keep it going in PINN due to its firmware downgrade facility. See https://github.com/raspberrypi/noobs/issues/520

Recalbox has just moved from github to gitlab and the location of the NOOBS files has changed, so the os_list needs updating. I will get onto this and then recalbox will be restored to PINN.

The good news is that this version also includes new firmware so it should be back in NOOBS soon as well.

ping @XECDesign for info ->https://gitlab.com/recalbox/recalbox/tree/master/updatesv2/noobs

procount, 2019-03-28:
ping @digitalLumberjack - I'm jumping the gun a bit. Just let me know when the new NOOBS files are ready :wink:

XECDesign, 2019-03-28:
As far as NOOBS goes, I don't think Recalbox will be added again.

daviduco, 2019-03-28:
> > i installed it without pinn
> 
> The file needed for standalone image-installation is very different to the multiple files needed for a PINN-installation of the same OS.

Offline install is the same

lurch, 2019-03-28:
:heart: at the amount of effort @procount puts in to making PINN compatible with various 3rd party OSes.

procount, 2019-03-29:
Thanks to Matt, a temporary copy of the latest version 6 of recalbox is now available through PINN.
This will be replaced by the official version when it is available.

procount, 2019-03-29:
solved.

closed 2019-03-29 by procount