procount/pinn #243: Void linux issues

aTosser, 2019-04-16:
The void linux version provided has a couple problems:

1. the non musl versio fails to install complaining about file integrity.
2. the musl version is so old that the package manager points to the incorrect repository location. (Void Linux no longer controls voidlinux.eu domain.
I assume that the standard void version would also have the same repo issue as the musl version. A new package needs to be built to facilitate this change and the standard version needs to have its installer looked at.

procount, 2019-04-16:
Thanks for informing me.
1. That'll be my fault. I will correct it asap.
2. I will look at updating all void versions.

procount, 2019-04-16:
Checksum is now fixed - a hangover from testing.

procount, 2019-04-16:
2018-11-11 seems to be the latest version now.
This is the result of converting these versions so far:

void1 - PiZero - OK
void1M- PiZero - not working
void2 - 3B+/2B - OK
void2M - 3B+/2B - not working - 4 flashes, missing start.elf?
void3 - 3B+ - OK
void3M - 3B+ - OK

win8linux, 2022-08-05:
Bumping this to mention that the following images are currently non-working and still use the defunct `voidlinux.eu` repos:
* void1
* void3

Haven't tested the musl images yet, but will soon.