procount/pinn #246: Possible to control PINN with an IR Remote via LIRC?

ailicic, 2019-04-16:
Is it possible to control PINN with an IR Remote via LIRC? I have an IR sensor on the GPIO that works in Kodi. Can I use something like lircrc/irexec in PINN to select an OS on boot?

procount, 2019-04-16:
I don't know much about LIRC.
If you can run some sort of LIRC daemon from the pinn_init.sh script that is able to decode the IR to standard keypress events (similar to how CEC works) then it may be possible for you to add it on yourself.
If it requires direct support in PINN, it may take me a while, but I could add it to the future roadmap.

ailicic, 2019-04-17:
I took a peek at what was available in PINN via ssh and I don't see any lirc processes running or any of the binaries available. As there is no package manager it would need to be baked in the PINN image. In prior years LIRC was installed as a package on linux machines but more recently some parts of it have moved into the kernel. See [here](https://kodi.wiki/view/HOW-TO:Set_up_LIRC#Changes_in_Linux_IR) for more. Is there anything more I could try on my end?

procount, 2019-04-17:
I guess I would have to build it into my kernel then if it won't work in just userland.

ailicic, 2019-04-17:
Ok. I'll leave that decision with you. Thanks for the great software and fast responses.