procount/pinn #247: "bsdtar: Pathname can't be converted from UTF-8 to current locale." error with custom os

mactooby, 2019-05-03:
I am following your guide (https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN) and os  successfully shown in the installation menu after booting PINN from SD card but during installation I got following error 

`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Linkname can't be converted from UTF-8 to current locale.`
`bsdtar: Linkname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Linkname can't be converted from UTF-8 to current locale.`
`bsdtar: Linkname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Pathname can't be converted from UTF-8 to current locale.`
`bsdtar: Error exit delayed from previous errors.`

I have used both Linux and Mac OS to create both boot.tar.xz and root.tar.xz files and same error shown. 

My locale settings as following:

`$ locale`
`LANG="en_US.UTF-8"`
`LC_COLLATE="en_US.UTF-8"`
`LC_CTYPE="en_US.UTF-8"`
`LC_MESSAGES="en_US.UTF-8"`
`LC_MONETARY="en_US.UTF-8"`
`LC_NUMERIC="en_US.UTF-8"`
`LC_TIME="en_US.UTF-8"`
`LC_ALL=`
I even added `LANG="en_US.UTF-8"` option to recovery.cmdline file and same error occurred. 
Even changing language and keyboard to us in PINN installation menu did not resolve the problem. 
 
Your help is appreciated.

procount, 2019-05-04:
I seem to recall I had a similar issue when I first changed to bsdtar. What version of bsdtar are you using?
Have a search in noobs issues to see if there are any clues/notes.
Failing that you could probably just use tar if you are creating the files for a fairly standard OS.

mactooby, 2019-05-04:
On Mac os I used "bsdtar 2.8.3" and on Linux I used "bsdtar 3.2.2". 
I have checked Noobs issues and googled around but no clear fix or solution. 
I would like to start with fresh copy of both os ( OSMS and Retropie )  but I spent good amount of time stabilizing them to work correctly and would like to multi boot them on same pi.

procount, 2019-05-04:
Try bsdtar 3.3.1 or later :shrug:

closed 2020-04-02 by procount