procount/pinn #250: allow ESC to exit PINN from all toolbars

drwhite729, 2019-05-08:
In the PINN user interface, when the Main Menu toolbar is showing, the ESC key exits PINN and boots an OS.

If instead, I type "m" or click "More" to go to one of the other toolbars (Archival or Maintenance), then ESC no longer works; I have to cycle back to the Main Menu before I can exit PINN.

Intuitively, I keep expecting ESC to exit PINN no matter which toolbar is showing.

lurch, 2019-05-08:
I wonder if using ESC as a "back button" might be more intuitive behaviour (if in a "sub menu") than immediately exiting PINN? :man_shrugging: 
(or maybe I'm simply misunderstanding the suggestion)

procount, 2019-05-08:
@lurch - (You really need to load it up and run it 😃). 
They're not sub-menus, but rather three toolbars that cycle around 1,2,3 and back again (an extension of the Advanced menu that was never implemented in NOOBS).

@drwhite729 - Yes I understand as I often find it a pain to cycle back to the main menu to exit myself.
I will consider it, but it takes up an extra button on the toolbar and they're getting a bit tight. I often wonder the opposite about the OS Info button, too.

lurch, 2019-05-08:
@procount Yeah I know, if I ever had more than 5 minutes of free time... :wink: 

Does Qt have the ability to have some kind of "global keypress handler" (that you could use to handle ESC), rather than specifically binding a specific keypress to a specific button?

procount, 2019-05-08:
Well yes, remember the Easter Egg? But maybe something that is restricted to the MainWindow would be better, probably along the lines of a keypress event filter that I use to cancel various timers.

lurch, 2019-05-08:
I was deliberately not mentioning that :egg: , but hoping that you would remember it too... :wink: :rofl:

procount, 2019-05-08:
Sorry 😢 Corrected.

lurch, 2019-05-08:
Doesn't matter, I was just teasing :stuck_out_tongue_winking_eye:

procount, 2019-06-08:
Will be implemented in the next version.

procount, 2019-06-12:
done

closed 2019-06-12 by procount