procount/pinn #251: copy relevant noobsconfig documentation into pinn documentation

drwhite729, 2019-05-08:
In the PINN documentation, when it gets to the section "How to customize an OS install", it doesn't give much detail; it just points the reader to the noobsconfig documentation.  I think it would be easier to follow if the relevant information was integrated into the PINN documentation instead.

As it is, both documents seem to say that noobsconfig is obsolete, that its functionality is integrated into PINN, and that not all of the noobsconfig steps are required in PINN.  However, it's difficult to tell just exactly which steps are required and which are not.  It's also hard to tell exactly where I need to put the customization files.

In my setup, I have an SD card and a USB stick.  Originally, the SD card only had a RECOVERY partition and the USB stick was empty, but after installing and reinstalling different flavors of Raspbian a few times, I now have:
- SD card
  - RECOVERY partition
    - /os
  - SETTINGS partition
    - /os
      - /Raspbian
      - /Raspbian_Lite
      - ...
- USB thumb drive
  - PINN partition
    - /os
      - /Raspbian
      - /Raspbian_Lite
      - ...

I'm having trouble with these questions:
- Where do I put the ssh or ssh.txt file so that, after I install some flavor of Raspbian and boot into it, SSH is enabled?
- Where do I put wpa_supplicant.conf so that PINN can access the internet through wifi?
- Where do I put wpa_supplicant.conf so that Raspbian can access the internet through wifi?
- Do I need to modify a partition_setup.sh file?  The documentation says I shouldn't, but none of my files have the line that calls customize.sh like the documentation shows.
  - Which partition_setup.sh do I need to change?  There's one in each directory in SD card / SETTINGS partition /os/Raspbian* and in USB drive / PINN partition /os/Raspbian*.
- Do I need customize.sh from noobsconfig?  Where do I put it?
- Where do I put retro.sh?  The noobsconfig documentation says it goes in the root of the RECOVERY partition, but that might not be accurate in PINN.
- If I want to customize multiple flavors of Raspbian (i.e., both Raspbian and Raspbian_Lite) with the same post-installation files, do I have to maintain multiple copies of those files (or tarball), or is it possible to have just one copy?  Where can I put it where it is accessible to all flavors?

procount, 2019-05-08:
I'm a little confused with your diagram. Did you boot the RPi from PINN on the SD card, and then install the OSes to the USB stick? In which case, there are some missing partitions on the SD card, because the Raspbian and Lite boot partitions will be there too - maybe you just missed them off?

Or did you mean "downloading" the OSes from the internet to your USB stick, from which you are then going to add custom features and install from USB stick to SD card.? (In which case you missed off the boor and root partitions for each OS on the SD card.)

> - Where do I put the ssh or ssh.txt file so that, after I install some flavor of Raspbian and boot into it, SSH is enabled?

On the PINN Recovery partition (where you will find recovery.rfs)

> - Where do I put wpa_supplicant.conf so that PINN can access the internet through wifi?

On the PINN Recovery partition (where you will find recovery.rfs)

> - Where do I put wpa_supplicant.conf so that Raspbian can access the internet through wifi? 

On the PINN Recovery partition (where you will find recovery.rfs) (+ Whatever you configure in PINN will get copied from PINN to Raspbian)

> - Do I need to modify a partition_setup.sh file? The documentation says I shouldn't, but none of my files have the line that calls customize.sh like the documentation shows.

No. That was only for Noobsconfig. PINN now includes the equivalent of cusotmize.sh

> - Which partition_setup.sh do I need to change? There's one in each directory in SD card / SETTINGS partition /os/Raspbian* and in USB drive / PINN partition /os/Raspbian*.

You don't need to anymore if you are customising a standard OS.

> - Do I need customize.sh from noobsconfig? Where do I put it?

No. It's built into PINN.

> - Where do I put retro.sh? The noobsconfig documentation says it goes in the root of the RECOVERY partition, but that might not be accurate in PINN.

It's the same. (I assume retro.sh still works, haven't tried it in a while).

> - If I want to customize multiple flavors of Raspbian (i.e., both Raspbian and Raspbian_Lite) with the same post-installation files, do I have to maintain multiple copies of those files (or tarball), or is it possible to have just one copy? Where can I put it where it is accessible to all flavors?

Originally, yes you would have to duplicate the custom files because PINN looks for these files in the same folder as the OS installation files. However, I have been enhancing this feature. There is now a configpath parameter that enables you to store them in a common folder, but I need to document it better.

Yes, you're right, the OS customisation documentation could do with a complete overhaul. I had hoped to get away with the noobsconfig documentation, but as I improve the functionality it is diverging more and more and needs integrating better. Perhaps I will start off by copying the documentation from noobsconfig to pinn then updating it. Integrating it directly into README_PINN.md might make it too big?, it's already getting quite large. You're welcome to submit a PR too if you think you can improve the documentation. All help gratefully received. 😉

drwhite729, 2019-05-08:
I'm booting from the SD card, and I've downloaded Raspbian OS's to the USB stick for customization and installation.  So the SD card has partitions RECOVERY, SETTINGS, boot, root, and data.  The USB stick has a partition named PINN -- but that's not PINN; that's just what I named it; it's to hold the downloaded Raspbian OS's that I intend to install.  Maybe I should have named it OS instead.

drwhite729, 2019-05-08:
OK, so the configpath parameter goes in the SD card, RECOVERY partition, recovery.cmdline file, and it specifies the directory where PINN can find (flavour)_(partitionName).tar, (flavour)_(partitionName).tar.xz, and (flavour)_(partitionName).txt, correct?  And where will PINN will look for this directory -- in the SD card's RECOVERY partition, or the USB drive's DOWNLOAD partition?  (I'm guessing the DOWNLOAD partition, since that's where the customizations would be without configpath, but I could be wrong.)

Assuming I'm using .tar files for the customization, how will the configpath parameter help me?  Won't I still have to have two copies of the tar file if I want to customize two different flavors?  (That is, I'll need Raspbian_root.tar and Raspbian_Lite_root.tar, won't I?)  It seems that configpath will only be useful if I'm using .txt files for the customizations -- I'll still need two of them (Raspbian_root.txt and Raspbian_Lite_root.txt), but they can at least refer to the same files.

procount, 2019-05-08:
Yes, you are correct about the location of configpath. But this specifies an absolute folder location, so you can reference any folder you wish:
/mnt/os/<config> for the SD card recovery partition
/tmp/media/sda1/os/<config> for the USB stick, or anywhere else that is accessible.

Another reason for using configpath was to add local customisations to an internet based remote OS, where it may not be possible to add your customisations into the remote URL.

If you're just using tar files, then configpath just allows you to locate them away from the OS installation files. Txt files will be necessary if you want to share the tar files. I added some additional features to include txt files within txt files, so you can have a set of customisation files for different features, and then just use an overall txt file to include the ones you want. I also added features to hopefully allow all this to happen for local and remote files, but it all needs testing and documenting. and more examples writing.

procount, 2019-05-08:
Here is brief description of the new features in the txt files:
```
@/filename.* = resets to original _srcFolder/filename.*  
@push= newFolder stack=_srcFolder, _srcFolder=newfolder  
@pop restores _srcFolder=stack  
@filename.* adds srcFolder  
@filename.txt - processed txt file as command file
@filename.tar - untarred 
@filename.tar.xz - untarred
Filename destination attributes user group - copies Filename to destination with properties
```
In many cases, filename can be a remote URL as well.
Each OS can also have its own configpath in its os.json.

drwhite729, 2019-05-09:
OK, I never could have guessed where those partitions would be mounted during the install process.  That definitely needs to be documented.  I might take you up on your offer to let me update your documentation for you and submit a pull request.  I can't be sure the documentation is correct, though, unless I can actually make it work like it's supposed to, and I'm still having trouble with that.  I tried to install Raspbian Lite, without a configpath, and get it to install a custom file in /etc/environment, and it didn't work.  Should I file a separate issue for that?

procount, 2019-05-09:
I don't think the retrospective customisation will work anymore. You will have to do it at install.
How abiut I have a fist stab at documenting the new behaviour then you can review it and submit changes?

procount, 2019-05-09:
I've added a very rough draft here -> https://github.com/procount/pinn/blob/master/noobsconfig.md
Probably lots of spelling errors, inconsistencies and lack of detail etc., but it's a start.

The basic noobsconfig concept still works, but many of the new features (Current Source Folder, use of URLs etc) have not been well tested, so I encourage you to test them out to destruction and feedback any problems, as well as testing out and improving the documentation. It would benefit from some good use case examples too. :wink:

drwhite729, 2019-05-19:
That link yields a 404 Page Not Found error.

procount, 2019-05-19:
You were a bit slow downloading :wink:
Sorry, I think it got lost in the translation updates I just did.
I've put it back again now.

procount, 2019-07-06:
@drwhite729  - Did you make any progress on this?

drwhite729, 2019-07-11:
Sorry, no.  My personal life went to hell for a while there.  I won't be able to commit to anything for some time.  In fact, I'm not even going to be using my Raspberry Pi anytime in the near future.  If you're keeping this issue open just for me, it's OK to close it.

procount, 2019-07-11:
Sorry to hear that.
I'll keep this open as a reminder that I need to finish it off sometime.