procount/pinn #255: PINN doesn't apply noobsconfig customizations to new installation of Raspbian

drwhite729, 2019-05-19:
I have a Raspberry Pi 3, an SD card, and a USB thumb drive.  PINN, Raspbian, and a data partition are installed on the SD card, and four different Raspbian flavors are downloaded (by PINN) to the USB drive.  PINN just updated itself today.
* The SD card contains:
  * a RECOVERY partition
  * a SETTINGS partition
  * a boot partition (mounted to /boot)
  * a root partition (mounted to /)
  * a data partition (mounted to /mnt/data)
* The USB drive contains:
  * a DOWNLOAD partition
    * /os
      * /Raspbian
      * /Raspbian_Full
      * /Raspbian_Lamp
      * /Raspbian_Lite

I want to replace my existing Raspbian installation with one from the DOWNLOAD partition, and I want PINN to automatically replace /etc/hostname and /etc/environment with my own files.  My custom hostname is currently saved in /etc/hostname, and I edited /etc/environment to contain the line that looks like this:
`export FLAVOR=lite`

I created these tar files:
* USB / DOWNLOAD /os/Raspbian/Raspbian_root.tar
* USB / DOWNLOAD /os/Rasbian_Full/Raspbian_Full_root.tar
* USB / DOWNLOAD /os/Rasbian_Lamp/Raspbian_Lamp_root.tar
* USB / DOWNLOAD /os/Raspbian_Lite/Raspbian_Lite_root.tar

with this command:
`tar cf Raspbian_root.tar /etc/environment /etc/hostname`

and I edited /etc/environment before creating each tar file so that every tar file would have different contents in its /etc/environment file.

Then I rebooted.  In PINN, I replaced my current Raspbian installation with one of the Raspbian flavors from my USB drive.  I've tried Raspbian and Raspbian Lite so far.  I did it in such a way that the data partition would be preserved.  When PINN finished installing the new flavor and booted it, my custom /etc/environment and /etc/hostname files were not copied to the new flavor of Raspbian.

procount, 2019-05-20:
That should have worked. Let me repeat it and I'll see if there is a problem.
Meantime, If you try again, you might want to look at the troubleshooting section of the wiki for details of how you can capture the debug log file. That might throw some light on the problem.

procount, 2019-05-20:
Ahh. Sorry, I've introduced a bug whilst adding the backup facilities that is preventing the customisations from executing. I'll release a new version soon to fix it.

procount, 2019-05-21:
I released a new version 3.2.3 which should now apply the customisations properly.

closed 2019-05-21 by procount

drwhite729, 2019-05-22:
It's working.  Thanks.