procount/pinn #256: PINN doesn't process customizations specified in a .txt file

drwhite729, 2019-05-23:
This is similar to issue #255, except this time, I have both a .tar file and a .txt file with customizations to apply.  The .tar file is being processed correctly, but the .txt file is not.

I have an SD card with PINN and Raspbian, and I have a USB drive with an os directory where I have downloaded four different flavors of Raspbian to install.  I've created these files (plus similar ones for the other flavors):
* USB / DOWNLOAD /os/Raspbian_Lite/Raspbian_Lite_root.tar
* USB / DOWNLOAD /os/Raspbian_Lite/Raspbian_Lite_root.txt

Raspbian_Lite_root.tar contains /etc/environment and /etc/hostname, and those are getting installed correctly when I install Raspbian lite.

Raspbian_Lite_root.txt contains:

    etc/rc.local                  /etc                 0775 root root
    etc/apt/apt.conf.d/zz-config  /etc/apt/apt.conf.d  0664 root root
    etc/dpkg/dpkg.cfg.d/zz-config /etc/dpkg/dpkg.cfg.d 0664 root root
    etc/skel/.ssh/authorized_keys /etc/skel/.ssh       0644 root root
    etc/skel/.bash_logout         /etc/skel            0664 root root
    etc/skel/.bashrc              /etc/skel            0664 root root
    etc/skel/.profile             /etc/skel            0664 root root
    etc/sudoers.d/zz-sudo-group   /etc/sudoers.d       0644 root root

I've also created these files:
* USB / DOWNLOAD /os/Raspbian_Lite/etc/rc.local
* USB / DOWNLOAD /os/Raspbian_Lite/etc/apt/apt.conf.d/zz-config
* USB / DOWNLOAD /os/Raspbian_Lite/etc/dpkg/dpkg.cfg.d/zz-config
* USB / DOWNLOAD /os/Raspbian_Lite/etc/skel/.ssh/authorized_keys
* USB / DOWNLOAD /os/Raspbian_Lite/etc/skel/.bash_logout
* USB / DOWNLOAD /os/Raspbian_Lite/etc/skel/.bashrc
* USB / DOWNLOAD /os/Raspbian_Lite/etc/skel/.profile
* USB / DOWNLOAD /os/Raspbian_Lite/etc/sudoers.d/zz-sudo-group

As far as I can tell, I'm following the directions correctly; everything seems to be where it is supposed to be.  However, after installing Raspbian Lite, none of the files specified in Raspbian_Lite_root.txt got copied to the new installation.  Again: Raspbian_Lite_root.tar is working; Raspbian_Lite_root.txt doesn't appear to be working.

procount, 2019-05-23:
Just to correct your terminology for clarity: These are not "flavours" as I use the term in PINN, but totally different OSes, with different names, different tar.xz files, and stored in their own folders.
"Flavours" are variations on the installation that all use the SAME OS, Same tar.xz files all stored in the same folder. Thus they share the tar.xz files and use a "flavours.json" file to create different variants of the same OS with different names, and thus use different customisations to make the final install different.

I just tried repeating this, using your post as an example and it worked fine. So it must be something you are doing wrong, this time.

Did you double check the filenames (including case)?
Did you look at the debug log to see if there were any errors?
If you have rebooted since this installation, the debug log will have been lost and you will have to repeat the installation.
Whilst it is installing, just drop into the recovery shell (ctrl-alt-f2) and login with username: root, password: raspberry, then type the following:  
`tail -f /tmp/debug`  
and watch the logs.
(You can toggle between the gui and the shell with ctrl-alt-f1 / ctrl-alt-f2)
At the end of the installation, look out for the line beginning `Checking for partition customisations
`

Here is the output when I did it. There are several errors about missing files, because I only included the zz-* files, and none of the files which looked like modifications of existing files (sorry I was being lazy) But the files I did include (sudoers.d.zz-blah-blah etc) did copy across fine:

```
Checking for partition customisations
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar | tee /tmp/fifo | bsdtar -xf - -C /mnt2  --no-same-owner  | sha512sum /tmp/fifo > /tmp/sha1.out.txt""
"cat: can't open '/tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar': No such file or directory
"
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar.xz | tee /tmp/fifo | xz -dc | bsdtar -xf - -C /mnt2  --no-same-owner  | sha512sum /tmp/fifo > /tmp/sha1.out.txt""
"cat: can't open '/tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar.xz': No such file or directory
xz: (stdin): File format not recognized
"
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_root.tar | tee /tmp/fifo | bsdtar -xf - -C /mnt2  | sha512sum /tmp/fifo > /tmp/sha1.out.txt""
Calculated Checksum= "7b410cd4547d3f3b9efb080de81ed712dda88fca547bb10fe8bf22e478259471a3a900430666a0993379282bad5fdf74cddce72a7850c28689cfecb28199eefd"
finished writing filesystem in 0.029 seconds
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_root.tar.xz | tee /tmp/fifo | xz -dc | bsdtar -xf - -C /mnt2  | sha512sum /tmp/fifo > /tmp/sha1.out.txt""
"cat: can't open '/tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_root.tar.xz': No such file or directory
xz: (stdin): File format not recognized
"
WARNING: Source file:  "/tmp/media/sda1/os/Raspbian_Lite/etc/rc.local"  does not exist
WARNING: Source file:  "/tmp/media/sda1/os/Raspbian_Lite/etc/skel/.ssh/authorized_keys"  does not exist
chmod: /mnt2/etc/skel/.ssh/authorized_keys: No such file or directory
chown: /mnt2/etc/skel/.ssh/authorized_keys: No such file or directory
WARNING: Source file:  "/tmp/media/sda1/os/Raspbian_Lite/etc/skel/.bash_logout"  does not exist
WARNING: Source file:  "/tmp/media/sda1/os/Raspbian_Lite/etc/skel/.bashrc"  does not exist
WARNING: Source file:  "/tmp/media/sda1/os/Raspbian_Lite/etc/skel/.profile"  does not exist
Checking for firmware Overrides

lurch, 2019-05-23:
Slightly OT...
```
"cat: can't open '/tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar.xz': No such file or directory
xz: (stdin): File format not recognized
```
Why not check that the file(s) exist, before trying to `cat` them?

procount, 2019-05-23:
Because.....,  this way it produced a nice error message.......? 😄

drwhite729, 2019-05-30:
I'm pretty sure the filenames had the right case.  I'll try again and see what the debug log shows.

drwhite729, 2019-05-30:
Abbreviated contents of the /tmp/debug file at the end of installation:

Checking for partition customisations
Executing: "sh -o pipefail -c "cat /tmp/media/sda1/os/Raspbian_Lite/Raspbian_Lite_boot.tar | ...
cat: can't open '/.../Raspbian_Lite_Boot.tar': No such file or directory
Executing: "sh -o pipefail -c "cat /.../Raspbian_Lite_boot.tar.xz | ...
cat: can't open '/.../Raspbian_Lite_boot.tar.xz': No such file or directory
Executing: "sh -o pipefail -c "cat /tmp/.../Raspbian_Lite_root.tar | ...
Calculated Checksum="...."
finished writing filesystem in 0.024 seconds
Executing: "sh -o pipefail -c "cat /.../Raspbian_Lite_root.tar.xz | ...
cat: can't open '/.../Raspbian_Lite_root.tar.xz': No such file or directory
Checking for firmware Overrides
Examining "/mnt/firmware.override/fixup.date" ...
Examining "/mnt/firmware.override/bootcode.bin ...
Examining "/mnt/firmware.override/start.elf"...

So, I can see where it's looking for *_boot.tar, *_boot.tar.xz, *_root.tar, and *_root.tar.xz, but I don't see it looking for *_boot.txt or *_root.txt.

drwhite729, 2019-05-30:
And yet -- this time, after installing and rebooting, even though the /tmp/debug file never mentioned it was processing the *_root.txt file -- all of my files got copied to my new installation of Raspbian Lite.  Go figure.  I swear it didn't work last week, but it's working now.  I don't know what I did differently, besides logging into the recovery shell during the installation process.  Oh well.  I'll do it again on the other versions (not flavours) of Raspbian, and if it doesn't break again, then I'll concede that it must have been user error on my part.

procount, 2019-06-08:
Can this be closed now?

closed 2019-06-11 by drwhite729