procount/pinn #258: Cannot install SolydX

Lonerider2010, 2019-05-24:
When I heard about SolydX, I wanted to install it on a Raspberry Pi.
As usual, I wanted to use PINN for installing the OS.
And, indeed, I found some traces in PINN for SolydX.
The file 'osGroupMap.json' contains an entry for 'solydx', telling me
that I should find this OS in the 'General' group.
Even in SourceForge, in Files and Home/os there is a directory 'solydx'.
But when I tried to install it with PINN, I did not find any entry
for SolydX, neither in the 'General' group nor anywhere else.
So my question is: How can I install SolydX on a Raspberry Pi
with the help of PINN?

procount, 2019-05-25:
Which model of RPi do you have?

Lonerider2010, 2019-05-26:
I have the Raspberry Pi 3 Model B+.

procount, 2019-05-26:
The version I distribute via PINN is from 2017 and is only suitable for use on the PI2 and Pi3B since it does not have the correct firmware for the 3B+.
I'm not sure what has happened with SolydX - I think it only has community editions now. If there is a later version available, I'll try and add it, but I'm rather busy with new features at the moment. If you can provide a link to a later version suitable for the 3B+, that might help.

procount, 2019-05-26:
I seem to have found a later version from 2018, which I think may work on the 3B+.
I'll try and convert it for PINN, but it might be a week before I can release it.

lurch, 2019-05-28:
> I seem to have found a later version from 2018...

Hmmm? https://solydxk.com/downloads/community-editions has a RPi version with a filename of `solydxrpi3_201903.img.gz` :wink:

procount, 2019-05-28:
I have that too but I think it is 64bit only.

⁣Get BlueMail for Android ​

Lonerider2010, 2019-05-28:
64bit is ok for the RPi 3 B+, isn't it?

procount, 2019-05-28:
I will try and convert all/both? of them when I can.

⁣Get BlueMail for Android ​

lurch, 2019-05-29:
Just a quick comment to say the amount of effort that @procount puts into converting and supporting various 3rd-party RPi OSes within PINN is really admirable! :bowing_man:

Lonerider2010, 2019-05-29:
Thumbs up also from my side :thumbsup:

procount, 2019-05-31:
Progress update.
It appears the 2018 and 2019 versions are both 64-bit only.
I've been using the 2018 version as a testbed. I've finally sorted converting it by hand, which will require some modifications to my conversion scripts. I think they will apply equally to the 2019 version which I will try next.

procount, 2019-06-01:
A new version of SolydXK 64-bit (2019-03-20) has need added to the General category.
This is suitable for the 3B, 3B+ and probably the 3A+ but I don't have one to confirm.
IIRC, it may be necessary to downgrade  PINN's firmware to make it boot on the 3B, but YMMV.

Lonerider2010, 2019-06-02:
Thanks a lot :smile: Will check that as soon as possible and report the results.

procount, 2019-06-03:
You don't need a new version of PINN. Just connect it to the internet and Solydxk64 will appear. 
PINN will tell you when a new version is available and then you can tell it to update.

Lonerider2010, 2019-06-04:
Now I could install SolydX via PINN without problems.
Thanks again:exclamation:

closed 2019-06-04 by procount