procount/pinn #263: [Feature] Add OpenBSD to OS List

jkhjhhbx, 2019-06-03:
**Feature**: Support OpenBSD
**Why?**: OpenBSD is currently a bit of a pain to install on the Raspberry Pi. 
Hopefully, this could help push more users to the wonderful OS.

**Related Links**:
https://www.openbsd.org/arm64.html

https://ftp.openbsd.org/pub/OpenBSD/6.5/arm64/INSTALL.arm64

https://ftp.openbsd.org/pub/OpenBSD/snapshots/arm64/INSTALL.arm64

___________________

Want to support this issue? Post a bounty on it! We accept bounties from [BountySource](https://www.wikipedia.org/wiki/Bountysource).

procount, 2019-06-03:
You're very welcome to have a go at converting it.
But you might want to have a read of the following issues first:
https://github.com/raspberrypi/noobs/issues/412
https://github.com/procount/pinn/issues/59

lurch, 2019-06-04:
@procount Perhaps this should be closed as a duplicate of #59 ?

procount, 2019-06-04:
Agreed

closed 2019-06-04 by procount

jkhjhhbx, 2019-06-04:
@procount @lurch #59 specifically states:

> FreeBSD (RPi3), RaspBSD, and NetBSD.

This issue is about OpenBSD, which is a whole other operating systems with its own complexities.

lurch, 2019-06-04:
@jkhjhhbx I had a read through some of the linked issues, and https://github.com/yzgyyang/freebsd-noobs/issues/1 is very informative - even the FreeBSD team themselves weren't able to get their OS working with NOOBS. So with no disrepect to @procount , I'd say the chance of him getting OpenBSD working in PINN is close to zero.

I suspect all the *BSDs will have similar boot/partitioning issues on NOOBS / PINN, so I think @procount should just add OpenBSD to the list in #59