procount/pinn #266: Please add Kali OS for Pi 3B+

Marineflieger, 2019-06-13:
I'm trying to add Kali OS for Raspberry Pi 3B+ as custom image to PINN (followed the noobs Guide) but don't get it working. The Installation hangs at 100% for over 30 Mins (sd card still working according to LED), and when I plug in/out power, Pinn acts like nothing is installed.

Thank you very much ( already for PINN)!

procount, 2019-06-14:
I have a number of OSes to update.
I'll try and get around to doing Kali soon.

kotpeepers, 2020-06-05:
@Marineflieger do you have instructions on how you did this? Im wanting to quad boot the pi 4 (4gb) and include Kali but I see that @procount does not have it in the list of OS to install with in PINN?
@procount is there any option in the future for Kali being added? I know when I used PINN on my Raspberry pi 3 that Kali was an option.
Thank You both for the assistance.

Dimensional, 2022-05-18:
I've downloaded and converted the latest release of Kali, 2022.2, to the PINN format. I've done a test boot for both of them on my Pi4, however this shouldn't affect their ability to be booted on their normally compatible hardware.
https://drive.protonmail.com/urls/MW24BX7WYR#2zZvtVkspIlx
https://drive.protonmail.com/urls/0P0TJARN0W#5gwZYbOAMdkD

Dimensional, 2022-05-19:
Kali has been doing Rolling Releases for a few years now, so technically any version just automatically gets updated to the latest whenever there's an update. However at one time I ran into an issue where an upgrade couldn't be installed, forcing me to download the newest at the time and convert it, losing any changes to my setup, so that's why I'm trying to get some of these new releases converted. I wish there was a faster way to get these done, however so far I can only make these conversions work by running Kali at least once on the Pi and then following the instructions.

procount, 2022-05-19:
It shouldn't be necessary to boot them before converting. I usually just loop mount the images. What problem do you have if you don't boot them?

Dimensional, 2022-05-19:
Last time I tried, it simply didn't boot when after I installed it using PINN. It's been a while, but maybe it needs some configuration to set up. Not to mention by following the instructions for custom distros, the ACL Permissions file ends up being blank if I use it before booting, but if I try to convert after an initial boot, the file needed in the instructions does populate.

I should point out that I somehow have to create the file first, then I can use the getfacl command to populate it. I get a permission error when trying to make it by redirecting the getfacl command to the file. Might be because I'm trying to do everything off of a Raspberry Pi OS installation.