procount/pinn #269: Add Raspbian Buster

mobluse, 2019-06-25:
I think these images should be added to PINN:
Raspbian Buster with desktop and recommended software
Raspbian Buster with desktop
Raspbian Buster Lite
https://www.raspberrypi.org/downloads/raspbian/

procount, 2019-06-26:
They are already there as PINN automatically lists all the same OSes as NOOBS.
But you must connect your Pi running PINN to the internet to be able to download them.

procount, 2019-06-26:
Can you see those OSes now?
There was a little delay in them being availble in PINN and NOOBS due to RPT not updating one of their files quickly enough, but that was resolved on Monday afternoon.
Can I close this issue?

closed 2019-07-11 by procount

cyklee, 2019-11-25:
@procount you may want to update the [supported OS page](https://github.com/procount/pinn/wiki/OSes-supported-by-PINN) to change Raspian Stretch to Buster if NOOBS has updated the distro versions.