procount/pinn #273: Localization - Proposals for typo fixes

bovirus, 2019-06-28:
About the current source file I have some proposals about fixes

Line 246

"Would you like PINN to create one for you?
WARNING: This will erase all data on your SD card"

It's multi-paragraph and should end with"." Like:

"Would you like PINN to create one for you?
WARNING: This will erase all data on your SD card."

It's the same for 

- Line 774
"OS(es) Downloaded with errors.
See debug log for details"
"OS(es) Downloaded with errors.
See debug log for details."

- Line 781
"OS(es) Installed with errors.
See debug log for details"
"OS(es) Installed with errors.
See debug log for details."

- Line 1792
"Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions"
"Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions."

and

- Line 1740/1786/1852
Change non-breakable space with standard space.

- Line 1847
This line seem available twice
"To change the background colour for the screen and most windows, add background=r,g,b, where 0,0,0 is black and 255,255,255 is white. I suggest to use values above 127."

lurch, 2019-06-28:
I'm not familiar with QT Linguist (been many years since I last looked at this translation stuff!) but I wonder if
"OS(es) Downloaded with errors.
See debug log for details."
and
"OS(es) Installed with errors.
See debug log for details."

could be split into 3 separate lines:
"OS(es) Downloaded with errors."
"OS(es) Installed with errors."
"See debug log for details."
and then the translator would only need to translate the string "See debug log for details." once, rather than multiple times? :wink:

bovirus, 2019-06-28:
@lurch 
There are separate string

- String1
"OS(es) Downloaded with errors.
See debug log for details"

-String2
"OS(es) Installed with errors.
See debug log for details"

Each string has a specific reference.
For me both strings should have "." at the end of string (because it's a multi-line/paragraph string and has a "." in the middle.

lurch, 2019-06-28:
> There are separate string

Yes, I realise that. I was just thinking-out-loud whether it would be possible to split the two strings into 3 separate (smaller) strings, as that _might_ ease the translation burden. :man_shrugging:  :man_shrugging:

bovirus, 2019-07-03:
@procount 

In multi-paragraph please add a  "." at the end of a multi-paragraph string. Ex

<source>Would you like PINN to create one for you?
WARNING: This will erase all data on your SD card</source>
<source>Would you like PINN to create one for you?
WARNING: This will erase all data on your SD card**.**</source>

<source>OS(es) Downloaded with errors.
See debug log for details</source>
<source>OS(es) Downloaded with errors.
See debug log for details.</source>

<source>OS(es) Installed with errors.
See debug log for details</source>
<source>OS(es) Installed with errors.
See debug log for details**.**</source>

<source>Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions</source>
<source>Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions**.**</source>

Thanks.

procount, 2019-07-03:
Thanks. I've sorted those in a separate commit but had problems with laptop so I'll push them up separately.

procount, 2019-07-08:
Implemented

closed 2019-07-08 by procount