procount/pinn #277: PINN 3.3 does not work on Raspberry Pi 4

Lonerider2010, 2019-06-30:
Tried to run PINN 3.3 on Raspberry Pi 4, but does not work.
Did the following steps:
- Downloaded PINN 3.3 from SourceForge
- Extracted the ZIP file
- Copied the content on FAT32 formatted SanDisk Ultra 32 GB SDHC card
- Placed in the Raspi
- Attached monitor via HDMI, and mouse and keyboard via USB
- Attached the power
- The red LED turned on
- The green LED blinked for a short time, then turned off
- Nothing else happened.

WallyCZ, 2019-06-30:
I have exactly the same problem. Pi 4B 4GB. Noobs 3.1 works without any problem.

Lonerider2010, 2019-06-30:
Also no problem with Noobs lite 3.1.

procount, 2019-06-30:
Ok. That's odd cos it did work fine. I'll check it out

procount, 2019-07-01:
Sorry. It seems my last deployment left out a critical file.
Please can you re-download pinn-lite.zip and try again.
Please let me know if this fixes it.

WallyCZ, 2019-07-01:
Yep, now it is working fine.

procount, 2019-07-01:
Thanks for confirming.
I added an announcement to my thread and will release a new version to allow self-update to fix this issue automatically soon. For now, a manual download or forced update is required.

closed 2019-07-01 by procount