procount/pinn #279: question: is data partition size configurable?

drwhite729, 2019-07-06:
When I use NOOBS or PINN to create an empty data partition along with Raspbian, the data partition is 512 MB.  Is that configurable?  I have an 8 GB SD card, so there's plenty of room to make that data partition a full 1 GB, but resizing existing partitions can be a pain.

procount, 2019-07-06:
Yes and no.
No, because it cannot be set by PINN's GUI.
Yes, but you need to download the data partition meta files to your local USB stick. Then you can edit partitions.json and set whatever size you like.
Or you can specify the sizes of each of your OSes (including the data partition) through pinn.mjh.nz

drwhite729, 2019-07-06:
Ah, yes.  Thanks.  Downloading it first didn't occur to me; that would work.

closed 2019-07-06 by drwhite729