procount/pinn #280: replace 1 OS with 2 OS's?

drwhite729, 2019-07-06:
I like that I can use the "replace" feature to reinstall an operating system without wiping out the data partition.  However, if I have 1 operating system plus a data partition, and I want to wipe out that OS and replace it with 2 operating systems, is there a way to do that while preserving the data in the data partition?

I don't need this badly enough to make it a request for a new feature -- I'm just curious if there's already a way to do it.

procount, 2019-07-06:
No. You'd have to reserve some initial space for the 2nd OS using a projectspace.
The only other way is to do it manually by resizing the partitions using gpartedor something, then editing the installed_os.json file yourself.

drwhite729, 2019-07-11:
ok, thanks

closed 2019-07-11 by drwhite729