procount/pinn #281: rpi4 doesnt boot. Using 7" TSDisplay

puccaso, 2019-07-07:
I'm using a 32gb sd card
with pinn installed as always
i'm using the 7" Touch Screen display with a rpi4 4gb
but I get no boot
nothing but a red led.
any ideas?

:(

procount, 2019-07-09:
> with pinn installed as always

Suggests you might have other RPi models? Have you tried using another of your RPIs to test out the SD card installation?
Are you using the latest version 3.3? Earlier versions will not work with the Pi4. 
Even if it is v3.3, please try downloading again to ensure you are not suffering form an early version error (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&start=775#p1489983)

I haven't tried the Touchscreen with a Pi4 because my ribbon cable cracked, but I have another on order so I will be able to test it soon.

dillasyx, 2019-07-24:
almost same problem here, too

I tried to use my 3B+ SD Card on the 4, but nothing booted, just red LED
it's PINN Lite 3.3

Then I tried a fresh installation of PINN Lite 3.3 on another SD Card and I got the colorfull bootscreen and then a bootloop and nothing else then the magenta, yellow, blue, cyan screen showed up

Then I used Etcher to flash the etcher image onto the SD card and now it stucks at the color screen.

When I use this SD Card in the 3B+ it works fine and shows v3.3

procount, 2019-07-24:
Released new v3.3c version of PINN.
Tested with the 7" touchscreen and it works fine, although it is best to add `disable_overscan=1` to remove the border, otherwise the touchscreen appears a bit off.
Download progress bar is also fixed.

closed 2019-07-31 by procount