procount/pinn #283: Lineage OS (KonstaKANG) 7.1.2 Nougat for RPI3 Boots to blank screen

prodigy213, 2019-07-09:
Hello,

Lineage OS (KonstaKANG) 7.1.2 Nougat for RPI3 boots to a blank screen. Once the screen is "loaded" (by waiting for the led light to stop flashing after OS selection) pressing Ctrl-Alt-F2 to get to the recovery shell does nothing. Ctrl+F[X] does nothing as well. Please let me know how I can assist you if needed.

Thank you

RaspberryPiFan, 2019-07-09:
I had the same problem without pinn. I used a USB stick. My fix: use a sd card for lineage os, remove all usb sticks.

procount, 2019-07-09:
Which model of RPi are you installing Lineage to?

prodigy213, 2019-07-09:
I'm sorry about that. I'm using Raspberry Pi 3 Ver. 2 Model B

prodigy213, 2019-07-12:
Today I did a reinstall on my family Raspberry Pi 3 Ver. 2 Model B to test Lineage, Window IoT and Screenly.I don't know if this helps but Lineage "marketing" stuff is showing RTAndroid information during the OS install.

procount, 2019-07-12:
Thanks. I'll look at is. Still working on lineage. Not working for me either now. Odd.

procount, 2019-07-12:
Temporarily removed Lineage until I can sort it.

peret2000, 2019-12-09:
Is there any update on this? Lineage is a nice android system. I am using it using a standalone SD (without PINN), so I have to constantly switch the rapsberry off, exchange the SD, and switch it on to switch between my Raspbian and Lineagee. I tried to install Lineage with PINN, looking at a previous version of  os_list_v3.json, and updating recovery.cmdline with alt_image_source directive. I managed to download and install Lineage 7.1.2, but if, actually, did not start: the screen goes black.
I would appreciate if there is some workaround. Maybe it is a probleam of Lineage. I am using Lineage 16 (Android Pie) in my 'standalone' SD, and works nicely.
Let me know if I can help in anyway, I am not an expert at all (I don't almost know anything about android), but I like to learn. Thanks!

procount, 2019-12-09:
No update, sorry. I will try to bump it up my priority list.

peret2000, 2020-03-31:
It is now possible, isn't it? I have been able to install Lineage 16 along with Raspbian Lite. Thanks a lot. It works fine, although it seems to be slower using PINN. I mean, I have imported my raspbian lite and Lineage from previous installations, following [the procedure indicated in your wiki](https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN); BTW, that procedure is brilliant, and very well explained.
Lineage seems to be slightly slower than the original installation. There can be many factors, of course, starting with the SD card, for example, that is not the same as the original Lineage installation.
If it is just the case that there are some tricks to optimise Lineage on PINN, it woud be a good news.
Regards, and good work!

procount, 2020-03-31:
The version of LineageOS installed by PINN is identical to one installed by eg  Etcher.
I only have to fix up partition references due to it being relocated. So performance should be identical.

Working on Feb version which should be out shortly.

lurch, 2020-03-31:
Related to #360 ?

procount, 2020-04-02:
LineageOS is now updated to the Feb 2020 release, with the TWRP recovery package now working too.

Konsta does comment on the graphics speed using SwiftShader and Mesa in his [FAQ](https://konstakang.com/devices/rpi3/LineageOS16.0/) so it may be slower than older versions.

closed 2020-04-02 by procount

lurch, 2020-04-06:
@procount I see that the page you link to says "Do not mirror my builds! Please post a link to this page instead.". Have you been in touch with the author to check that he's okay with your "repackaging" of his work? Maybe if you ask him nicely he'll add a link to your PINN project? :slightly_smiling_face:

procount, 2020-04-06:
This is about the third release of LineageOS I have made and I did ask Konsta about that statement before I released the first one, so he's ok with it. I already post links on his LineageOS thread on the forum.

Tomoli75, 2020-04-16:
Uhm, I still have this issue on the raspberry pi 3b+ with official touchscreen.

procount, 2020-04-16:
Are you using the latest version available from PINN (16.0 from Feb 2020)?
I will give it a try later.

Tomoli75, 2020-04-16:
Indeed I am, although the latest version installed from the Lineage OS Pi version (I apologise, can't remember the name nor link) doesn't work either, although other operating systems do.

procount, 2020-04-17:
@Tomoli75 - Did you follow the instructions in Konsta's FAQ about configuring LineageOS for the 7" touchscreen?

> Q: I have official 7” LCD display and touchscreen. What should I do?
> A: Official 7” touchscreen is only supported using SwiftShader software renderer. See below how to switch between different graphics drivers. You will also need to change display size under Settings -> Display -> Display size (or change ro.sf.lcd_density to 120 in /system/build.prop) to adapt to the smaller resolution.
> 
> Q: I need to use SwiftShader software renderer to use the official 7” display or I want to boot without any display connected. What should I do?
> A: Warning, SwiftShader is a software renderer and using it affects graphics performance. You can switch between MESA and SwiftShader graphics drivers by executing following commands in ‘adb shell’/serial console/terminal (you can enable built-in terminal app from Settings -> System -> Developer options -> Local terminal):
> 
> su
> rpi3-graphics.sh swiftshader
> su
> rpi3-graphics.sh mesa

lurch, 2020-04-17:
I've never used Android on Pi, but I assume the above FAQ entries imply that you need some _other_ display attached to your Pi, in order to be able to switch LineageOS over to displaying on the touchscreen?

Tomoli75, 2020-04-17:
I do have a HDMI monitor I can use, although there is no output to it?
(I'll attempt to add the ignore_lcd flag into config.txt and test from there, thanks for your help this far)

Tomoli75, 2020-04-17:
Just a quick update: It's working on a HDMI display.

procount, 2020-04-17:
I can confirm that changing the graphics driver to swiftshader et al does allow it to work on the 7" touchscreen with the Pi3. ( whereas the Pi4 works ok and does not need this change of drivers). It is a bit slower so you have to be patient with it.

Maybe I will have to add another "flavour" so that it can work with the touchscreen out of the box.

ukxisle, 2020-10-24:
how do i enable the swiftshader driver in PI 4 to use original 7" screen - all I get is a blank screen when booting lineage 17.1

procount, 2020-10-24:
Have you got Lineage 17.1 installed via PINN or is it just a normal image installation?

ukxisle, 2020-10-27:
Normal image installation. Maybe faulty hdmi vga adapter so purchasing
another also getting a new power supply
___________________________

Terry Keenor
____________________________
"Life is not an academic exercise"