procount/pinn #284: Translation - pelase check source and strings (end .)

bovirus, 2019-07-09:
@procount 

Coudl you pelase check source file and language file about these string

- File recovery\initdrivethread.ccp

Source - tr("Would you like PINN to create one for you?\nWARNING: This will erase all data on your SD card.")
language file - Would you like PINN to create one for you? WARNING: This will erase all data on your SD card

- File recovery/mainwindows.cpp

Source - "OS(es) Downloaded with errors.\nSee debug log for details."
Language file - "OS(es) Downloaded with errors.\nSee debug log for details"

Source - "OS(es) Backed up with errors.\nSee debug log for details."
Language file - OS(es) Backed up with errors.\nSee debug log for details"

- File recovery/optiondaialog.ui

Source - Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions.
Language file - Enables an SSH connection to the rescueshell in PINN. It cannot be used to control PINN, but having a shell is very useful to perform remote maintenance on any of the SD card partitions

In source seems that tehse strings end with "." and in language file not end with" .".

procount, 2019-07-09:
Thanks. I thought I had caught them all. At least the list is getting shorter :wink:

bovirus, 2019-07-09:
@procount 

Now it's strange because in the past the problem was the strings in source code.
Now the source code strings are right. Please don't change string text in the source.

I believe that the problem is in the process that make the translation_xx.ts from source code.

I get it because in Italian I have a "." at the end of multi-paragraph strings and QT linguist with end punctation check report this warning as with yellow ticks.

procount, 2019-07-09:
I had done all the edits, but because every edit to the source tends to modify the .ts files (all the line numbers change, for example), the .ts files are not automatically updated after each build. 
I added a new separate step to the build process to update them when it is time to release, but I need to remember to do it!

lurch, 2019-07-09:
> but I need to remember to do it

...or set up a pre-commit hook to do it automatically :wink:  https://github.com/raspberrypi/noobs#addingupdating-translations

bovirus, 2019-07-09:
@procount 

I believe that to avoid it I suggest to make a rebuild of .ts file every time that you make source changes.
For me the translation revision is not so a big problem (using QT Linguist).
Using QT Linguist (preferred tool for .ts language file) you can easily update and revise language file.
Standard editor (like Notepad++) is not so functional as QT Linguist to edit .TS file.
QT linguist also make check on end punctation, abbreviation, etc.

procount, 2019-07-09:
Hopefully the last commit I just pushed is ok now. But there is still one unfinished/missing translation for you to do.

procount, 2019-07-09:
> ...or set up a pre-commit hook to do it automatically 😉

When I first started, that was all black magic to me, so I didn't use it. And even now it might be a bit too often. I might try and put it in my deployment script, though, now that I have created a complete new build environment VM on ubuntu 18.04.

bovirus, 2019-07-09:
@procount 
Thanks for teh fixes

About this string

" When used in conjunction with forcetrigger&#xa0;it sets a maximum timeout period in seconds for the recovery menu to be displayed before continuing to the boot selection dialog and potentially selecting a default OS to boot."

There is a unbreakable blank space after forcetrigger.
What's about starting blank space? Is it need?

procount, 2019-07-09:
Hopefully fixed now.
Be aware that when this is FINALLY complete, I will more than likely rebase this branch and squash all these intermediate commits.

bovirus, 2019-07-10:
@procount 

New PR about Italian language.

In latest commit that you posted seems that there is a duplicated string

"When used in conjunction with forcetrigger it sets a maximum timeout period in seconds for the recovery menu to be displayed before continuing to the boot selection dialog and potentially selecting a default OS to boot".

Thanks.

bovirus, 2019-07-10:
Please check these strings that started with a space as first character (is it's right)

Selects all OSes where the most recent one is on a local USB drive
Waits for all OSes on local USBs to be listed
Waits for OSes from all sources to be listed
Similar to alt_image_source but it matches the NOOBS method of specifying a list of additional respositories as a space separated quoted string.
This setting allows the network timeout to be specified in seconds, especially for networks that take longer to establish.
setup script needs to be updated
partions_setup.sh script error
partition_setup.sh script OK
:BTRFS file format

There are also others strings that start with a blank space but "probably" are a sub-string.

To avoid problem like this I prooose you to avoid as much as possible to sue strings as result of merge of more string text.

ex.

 MB of backup space, but only

procount, 2019-07-10:
Done, but the last 4 are substrings.
Substrings are part of a bigger issue that I need to tackle to reformat them as whole strings + arguments, as Lurch previously pointed out. So you can ignore those for now as I don't have time to do them at the moment.

bovirus, 2019-07-10:
@procount 

Thanks.
Please add these job (use unique strings and not composite strings) to the wish list. :-)

lurch, 2019-07-10:
> I need to tackle to reformat them as whole strings + arguments, as Lurch previously pointed out

I actually started fixing this on Sunday afternoon :)  If / when I get some more free time I'll see if I can finish it off.

procount, 2019-07-10:
Ah, thanks! That's kind of you. :Smile:
Didn't you have anything better to do, like replying to emails :wink:

bovirus, 2019-07-12:
@procount 

Could be change

"Please accept these T&Cs to continue."
in
"Please accept these T & Cs to continue."

with a space after "&" you avoid use it as accelerator.
 
and 

"Unknown compression format file extension. Expecting .lzo, .gz, .xz, .bz2 or .zip"
in 
"Unknown compression format file extension. Expecting .lzo, .gz, .xz, .bz2 or .zip."

It's multi-paragraph with a "." in teh middle and it's need also "." at the end.

procount, 2019-07-12:
done & updated IT & ZH_TW.