procount/pinn #287: recovery.pro and translation_ae.ts/translation_ast.ts

bovirus, 2019-07-10:
@procount 
Using the QT SDK (instaleld v. 5.13.0) command

**_lupdate recovery.pro_**

in recovery folders I got this error

**_File D:/Download/pinn-master/recovery/translation_ae.ts won't be updated: it contains translation but the target language is not recognized
File D:/Download/pinn-master/recovery/translation_ast.ts won't be updated: it contains translation but the target language is not recognized_**

For the other language no error.

procount, 2019-07-10:
We're still on Qt 4.8 if that makes a difference?

bovirus, 2019-07-10:
Ok. I will check with QT SDK 4.8.

bovirus, 2019-07-10:
@procount 
It's true with QT 4.8.7 (mingw) I didn't get this error.

Running lupdate recovery.pro i got this error

C:/Qt4.8.7/mkspecs/features/device_config.prf(13): 

Querying unknown property CROSS_COMPILEProject MESSAGE: Disabling CEC support for Raspberry Pi - rpi-userland headers not found at 
../../staging/usrD:/Download/pinn-master/recovery/recovery.pro(25): 
Function 'system' is not implemented

Are you using QT 4.8.7 for mingw?

procount, 2019-07-10:
No, Qt4.8 on Ubuntu 18.04.
You can ignore that error message about CEC.

bovirus, 2019-07-10:
@procount 

OK. Then please close this PR.

I added a new PR about languages. Found untranslated string not reported by QT Linguist.

https://github.com/procount/pinn/pull/289

And please close this PR (not necessary).

https://github.com/procount/pinn/pull/285

closed 2019-07-10 by procount