procount/pinn #291: PINN Language update

bovirus, 2019-07-11:
Hello.

There are some translation that are not complete

Basque language - @aldata
Chinese (Taiwan) - @s8321414
French - @Jey2k
Korean - @sople1
Norwegian - @Øistein Rian
Portuguese - @igorruckert
Spanish - @dramosr 

Could you pelase complete if you can the relative translations?

The current status of translation is

** Statistics about 'translation_ae.po' - converted by 'translation_ae.po'
214 translated messages, 3 fuzzy translations, 258 untranslated messages.

** Statistics about 'translation_ast.po' - converted by 'translation_ast.po'
103 translated messages, 16 fuzzy translations, 356 untranslated messages.

** Statistics about 'translation_ca.po' - converted by 'translation_ca.po'
34 translated messages, 2 fuzzy translations, 439 untranslated messages.

** Statistics about 'translation_de.po' - converted by 'translation_de.po'
232 translated messages, 23 fuzzy translations, 220 untranslated messages.

** Statistics about 'translation_en.po' - converted by 'translation_en.po'
0 translated messages, 475 untranslated messages.

** Statistics about 'translation_es.po' - converted by 'translation_es.po'
162 translated messages, 54 fuzzy translations, 259 untranslated messages.

** Statistics about 'translation_eu.po' - converted by 'translation_eu.po'
198 translated messages, 17 fuzzy translations, 260 untranslated messages.

** Statistics about 'translation_fi.po' - converted by 'translation_fi.po'
43 translated messages, 11 fuzzy translations, 421 untranslated messages.

** Statistics about 'translation_fr.po' - converted by 'translation_fr.po'
187 translated messages, 17 fuzzy translations, 271 untranslated messages.

** Statistics about 'translation_hu.po' - converted by 'translation_hu.po'
100 translated messages, 16 fuzzy translations, 359 untranslated messages.

** Statistics about 'translation_it.po' - converted by 'translation_it.po'
475 translated messages.

** Statistics about 'translation_ja.po' - converted by 'translation_ja.po'
100 translated messages, 16 fuzzy translations, 359 untranslated messages.

** Statistics about 'translation_ko.po' - converted by 'translation_ko.po'
320 translated messages, 5 fuzzy translations, 150 untranslated messages.

** Statistics about 'translation_nl.po' - converted by 'translation_nl.po'
121 translated messages, 15 fuzzy translations, 339 untranslated messages.

** Statistics about 'translation_no.po' - converted by 'translation_no.po'
200 translated messages, 17 fuzzy translations, 258 untranslated messages.

** Statistics about 'translation_pl.po' - converted by 'translation_pl.po'
100 translated messages, 16 fuzzy translations, 359 untranslated messages.

** Statistics about 'translation_pt.po' - converted by 'translation_pt.po'
200 translated messages, 17 fuzzy translations, 258 untranslated messages.

** Statistics about 'translation_ru.po' - converted by 'translation_ru.po'
100 translated messages, 16 fuzzy translations, 359 untranslated messages.

** Statistics about 'translation_sv.po' - converted by 'translation_sv.po'
104 translated messages, 16 fuzzy translations, 355 untranslated messages.

** Statistics about 'translation_tr.po' - converted by 'translation_tr.po'
125 translated messages, 19 fuzzy translations, 331 untranslated messages.

** Statistics about 'translation_zh_TW.po' - converted by 'translation_zh_TW.po'
464 translated messages, 11 fuzzy translations.

Thanks.

bovirus, 2019-07-12:
Thanks to

@sople1 (Korean)
@s8321414 Chinese (Taiwan)

the relative langaueg is no complete.

UI hope that could be possible also for other languages.

lurch, 2019-07-12:
@bovirus With the tweaks that I'm making to the PINN source files, many of the English strings will be changing soon, so it's probably worth waiting a while before requesting other translators to update their translations... :slightly_smiling_face:

bovirus, 2019-07-12:
@lurch 
Please take

1. Most of the changes about end punctation mismatch in source English strings are already fixed. Then most the current translation didn't require in the future just few checks.
2. When apply the changes on source code strings with QTLinguist you can check easily the yellow tick to see the changes about source code and translated strings that have to validate.

lurch, 2019-07-12:
I'll only be changing the English (source) strings - I don't know any other languages well enough to attempt to update any of the translations.

bovirus, 2019-07-12:
@procount

Add new PR https://github.com/procount/pinn/pull/296 to fix two line breaks missing (vs source code).

Please close/ignore this PR https://github.com/procount/pinn/pull/295

Pelase close this PR https://github.com/procount/pinn/pull/290

lurch, 2019-07-19:
@bovirus Just to keep you up to date: I've made a whole bunch of translation related source-code changes, which will result in some extra strings needing to be translated in all languages.
I've not made a PR yet as I still need to tidy it up a bit, and I've been very busy all this week. And there's some weird build / SSL issue which myself and @procount are trying to diagnose.

bovirus, 2019-07-19:
@lurch 

Thanks for the changes. Please when you can update it as PR.
Then when it will be available we started to revise the translation.

procount, 2019-08-29:
Hey @lurch, how's progress on the translation arguments? I may need to regress to fix the PAL issue, so I'd like to include your fixes before I do.

lurch, 2019-08-30:
I've been **very** busy recently, but I'm hopefully gonna have some free time this weekend to finish off my PINN changes :crossed_fingers: :slightly_smiling_face:

procount, 2020-02-07:
@lurch - How are your translation changes getting along?
If you're still very busy, you could always post what you have and I'll continue the changes, as I need to look at some language issues soon, anyway.

lurch, 2020-02-07:
Sure thing, I'll try to sort out what I have (and maybe email it to you) on Sunday.
(although I'll hopefully be slightly less busy soon)