procount/pinn #304: Wiki: missing RPi4

RaspberryPiFan, 2019-07-20:
OSes supported by PINN and Model Names don't contain RPi 4

procount, 2019-07-20:
Which OSes, for example?
Only Raspbian libreelec and raspex support rpi4 so far.

lurch, 2019-07-21:
I think @RaspberryPiFan meant there's no Pi4 info on https://github.com/procount/pinn/wiki/OSes-supported-by-PINN or https://github.com/procount/pinn/wiki/Model-Names ?

procount, 2019-07-21:
Oh yes.
I need to update those.

lurch, 2019-07-22:
Why was this closed? :confused:

procount, 2019-07-22:
Reopening to remind me to do it.

procount, 2019-07-24:
First update done .Shame I can't shrink the text in the table.

closed 2019-07-24 by procount

lurch, 2019-07-25:
Looks like you could put a space in "ubuntuMate1604LTS" which might shrink down the first column? And given that the Pi1 and the Pi0 use the same SoC, maybe they could be merged into a single column?

procount, 2019-07-25:
It's a bit better, but I wonder if I just messed up my easy creation of this table from Matt's webpage 🤷‍♂

lurch, 2019-07-25:
Sorry, I didn't realise it was autogenerated. Maybe move it from the wiki onto a gh-pages site? (which I guess would then allow you to have horizontally-scrollable tables) :man_shrugging:

procount, 2019-07-25:
It's ok. I'd forgotten too, and have forgotten how I did it. Needs improvement (future task)