procount/pinn #305: OS Downloader remains at 0%

Dimensional, 2019-07-22:
The OS Downloader bar doesn't update as it's downloading the files to a USB thumb drive. Not sure what the exact cause is, however I'm using a formatted 200 GB microSD on a Raspberry Pi 4 2GB model. The Thumb Drive is connected to the top USB3 port.

At this time, it appears to remain that way while downloading multiple OSs in a batch or individually, but only the bar and percentage indicator aren't changed, while it still the details, namely downloading and checking texts, are displaying.

procount, 2019-07-22:
Which PINN  version are you using?
What format did you use for your USB thumb drive and how big is it?

Dimensional, 2019-07-22:
I'm using the latest official release, V3.3, as this is my RPi 4, and it's the Pinn-lite zip that I'm using, since I wanted to download each OS independently and use less room for PINN. The USB thumb drive is formatted to exFAT, and it's a 64GB Sandisk Ultra USB 3.0

Dimensional, 2019-07-22:
I switched to a console screen, the ctrl alt f2 function, and am able to confirm the downloads are happening. Again the only thing that appears to be bugged is just the percentage and the download bar.

procount, 2019-07-22:
OK, just checking since some people managed to download a test version of v3.3b which doesn't work so well. I'll check to see what I can find.

Dimensional, 2019-07-22:
Okay, I've attempted to download an OS with the thumb drive connected to the USB 2 port, and it's still showing 0% with the bar not moving. Just for reference, I didn't have this problem when using the earlier versions of PINN on my RPi3B with the same thumb drive, so the only differences are the version of PINN and the model of the Pi.

Dimensional, 2019-07-23:
Running the installer without a usb drive attached doesn't have the same bug. Progress bar and percentage do move. I tested by installing Raspbian Full on the Pi 4, and it downloaded and installed with the progress bar moving.

procount, 2019-07-23:
Found it. Forgot to add some plumbing. Will be fixed in next release.

procount, 2019-07-24:
Fixed in v3.3c now released

closed 2019-07-24 by procount