procount/pinn #306: Install a new os not in the list

sebas77, 2019-07-22:
I need some instructions to install lakka on rpi4 from their image files which already comes as tar. I would like to retain the existing partitions and os and create a new one for lakka. I could per creare the partition myself but other than that I am totally noob

procount, 2019-07-22:
AFAIK, Lakka is working on Pi4 support and have some nightly builds available, but I've not seen an official release for it.
PINN already supports Lakka and the version will be updated to include Pi4 support when it becomes available.

sebas77, 2019-07-22:
Ok so you are suggesting to wait, thank you. Even so if I got it right, installing a new os will reset all the other partitions as well, is that right?

procount, 2019-07-22:
Yes, please wait. If you want to convert a nightly beta version for yourself, there are instructions in the wiki, and in README_PINN.md

Even though Lakka is supported in PINN, PINN only shows the OSes that are suitable for your version of RPi, unless you choose the "showall" option.

When PINN installs OSes, it shares out all of the SD card space between them, so there is no space left to add any more partitions later. PINN does not support adding more OSes to an SD card for this reason.
You can do it manually if you wish. The easiest way (but it will take a while) is to backup all your installed OSes to a USB stick. Then select all those backups along with the new OS you want and install them back to the SD card. 
If you have room, and foresee that you might want to add another OS later, add a ProjectSpace as well, which you can later replace with another (compatible) OS.

sebas77, 2019-07-22:
I actually

closed 2019-07-22 by sebas77