procount/pinn #310: display in Pal issue

Ekogreencity, 2019-08-02:
I'am using a raspberry 3 with Pinn, and it used to work perfectly in Pal with the version V3.2.4c.
I had to press Shift and "3" to force Pinn to display in PAL mode on my TV who doesn't have any HDMI connexion.
Since downloading the last version of PINN-lite, this seems not to be possible anymore.
I tried to change the recovery.cmdline with display=2,but nothing seems to work to display an image in PAL mode.
I know the problem is just in Pinn, the different OS used can be displayed in PAL.
Anybody has found the same issue trying to display in PAL mode with PINN ?

procount, 2019-08-02:
I've not knowingly changed this so will take a look.

procount, 2019-08-02:
Can you confirm that it works properly in version 3.3?
https://sourceforge.net/projects/pinn/files/archive/pinn-33.zip/download

Ekogreencity, 2019-08-02:
thanks for your answer,
just tested version 3.3 
Pal Mode is perfectly ok for PINN with this older version

procount, 2019-08-03:
OK. I suggest you stick with 3.3 for the moment, then. 
It may take me a while to track this one down as there were some major build changes after v3.3, which shouldn't have affected this, but obviously did.

Ekogreencity, 2019-08-04:
Ok, I will do like that.
If it can help, it's look like the Pal Mode is displayed with a bad configuration, like if it's in 60hz instead of 50hz for example.
Regards

procount, 2019-09-05:
I have been chasing this bug for a while now.
I think it is related to changes in the firmware related to the new Pi4 and Buster multiple display capability. This shouldn't affect RPI models prior to the Pi4, however, and I hope to be able to get it working again for the next release.

Please be aware, however, that on the PI4, switching the display output between HDMI and composite will no longer be possible through tvservice (which is used by NOOBS and PINN to switch modes). This can only be done via config.txt before boot, since it requires changes to the clock setup that cannot be done at runtime.

As a result, this feature will be deprecated in future versions of PINN for the Pi4, maybe for earlier models too. So you should consider creating a config.txt file and explicitly setting PAL mode in there. This should allow you to go back to using the later versions of PINN with composite output.

bastianlstrube, 2023-10-28:
Any news on this issue?

I'm on a Pi 3 and can't get composite to work either.

procount, 2023-10-30:
This seems quite tricky to fix, given the changes to the firmware.

I've just tried PINN v3.3 on a Pi3B and managed to get the composite display to work on there.
Apart from the Gentoo images from Sakaki, I could also see the list of OSes and I tested raspios_arm64_lite could be downloaded ok. 
I didn't try any others, but it's probably due to the LetsEncrypt certificate expiry. If this is the only issue it may be possible to update the certificates in PINN v3.3 to allow them to be downloaded. Were there any particular OSes you struggled with?

To get PINN 3.3 to work with composite, I added 'sdtv_mode=2' to PINN's  config.txt and 'forcetrigger' to recovery.cmdline.
The display didn't always come on after a cold boot, so the forcetrigger option ensures PINN's recovery menu was triggered.
Then I pressed 'Esc' to clear the error dialog from isshoni's Gentoo website (PINN v3.3. doesn't have the correct https certificates for this website) and then pressed '3' to select PAL mode. (It doesn't seem to remember this anymore).

However, I didn't manage to get composite working under Raspios with the KMS display driver. Maybe you will have more luck, or try the FKMS or legacy drivers. Maybe you will have to find older versions of OSes to get composite to work

I'm not sure the Pi5 will improve this situation.

So I can only suggest downgrading PINN to v3.3 as you did before. If you do this on an existing installation, remember to remove 'runinstaller' from recovery.cmdline before booting, otherwise PINN will wipe all your OSes.

bastianlstrube, 2023-10-31:
Alright, I understand it get's tricky to continue support, Thank you for checking.

I got 3.3 to boot with composite, but wanted to install LibreElec and Lakka or Retropie.
None of those OS had working links.

procount, 2023-11-01:
Did it complain when downloading the https files? This is most likely due to the certificate expiry I mentioned above. If that is the case, I can try modifying 3.3 to allow downloading of these files. It may take a little while as that version used a different development environment.
Meanwhile, I can suggest the following methods to install your OSes:

1. Install the latest PINN and install the OSes using that. If you don't have a HDMI monitor, you could add the 'forcetrigger vncshare ssh' options to recovery.cmdline and control PINN over VNC from a PC over the network. Then replace PINN with the older 3.3 version, being careful to remove the 'runinstaller' option before use.
2. Install the latest PINN and download the OSes you want to a USB stick (see README_PINN.md for full details about setting up your USB stick first). Then you can install the older PINN 3.3 version and install the OSes from the USB stick.

procount, 2023-11-02:
I managed to resurrect v3.3 and added the Let's Encrypt certificates to it.
Please try https://sourceforge.net/projects/pinn/files/archive/pinn-33nc.zip/download

bastianlstrube, 2023-11-04:
The Ctrl + 3 to get pal output doesn't work anymore.
just retested the earlier 3.3.zip file you linked to in this thread, but that one works.

I'll try your other recoomendation with installing through the newest PINN version later today.

Thank you for looking into this!