procount/pinn #311: Won't connect to Wifi

ZhuDaHai, 2019-08-23:
Raspberry Pi 4B
Latest download of PINN on 16GB SD Card
List of Wifi networks does not show 5Ghz Network, only 2Ghz
Cannot connect to 2G wifi network.
Select the network from the list, enter the password, verify password with 'Show Password', click on connect and the status bar just bounces back and forth.
Any helps/suggestions?
Thanks

ZhuDaHai, 2019-08-23:
BTW: It doesn't remember the Language settings between reboots...

procount, 2019-08-23:
Is https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1524079#p1524030 also your post?

So I assume you are using the onboard wifi of the pi 4B?
I have just repeated the tests on v3.3c and I have no problem connecting on 2GHz wifi to my mobile phone hotspot or to my home router. I have not tried 5GHz, but I will when I get the chance.

I also tried changing the language and it remembered it between reboots. Which language did you try (in case it is language specific, but I doubt it)?

ZhuDaHai, 2019-08-24:
English (US). Keeps reverting to English (UK).
I did finally get it to connect to my 5Ghz network using wpa_supplicant.conf in /boot. 
It didn't actually connect, though, when I rebooted. But at least that made that SSID appear in the list and I could then connect to it. The 2GHz network would sometimes connect and then immediately drop. I rebooted the Wifi Access Point, but that didn't help.

RaspberryPiFan, 2019-09-05:
I had the same problem, but after waiting a long time, it worked