procount/pinn #314: Everything is now GONE!

ZhuDaHai, 2019-08-24:
I guess this is just beyond me. 
I can't install a specific OS to a ProjectSpace. It ether says its incompatible number of partitions or the partitions are too small.
I had install Raspbian FULL and Raspbian Lite into two ProjectSpaces - one at a time through the Maintenance Menu.
But I could not install rtandroid or Lakka_RPI2 due to those above errors.
So I tried to install just rtandroid by selecting the ProjectSpace 4 on the Utility tab and then rtandroid on the General tab.
In then proceeded to try to install Raspbian FULL - AGAIN.
That wiped out all my changes and now everything is lost. Everything. No more ProjectSpaces, no more installed OSes. Nothing.
If you really MUST select ALL OSes and set up ALL ProjectSpaces ahead of time by looking into your Crystal Ball and Predicting EXACTLY what you will want to work on/test a year from now, then is just too limiting to be of any use to me.
Did I miss something important buried in the docs somewhere or is this just how it is?

P.S. One of the most critical things I was hoping to be able to do with this was install multiple copies of Raspbian Lite on the USB HD, but it looks like that just not possible if you have to select ALL OSes up front - as you can't select the same OS TWICE.
This is just frustrating and depressing.

ZhuDaHai, 2019-08-24:
OK. ProjectSpaces (from the Readme):

> They reserve OS space, so the choice of which OS to install or add can be made later on.

But, like I said above, I created 8 empty project spaces, but was only able to install Raspbian FULL and Raspbian Lite on the first two. 

When I tried to install rtandroid or Lakka_RPI2 on any of the other EMPTY ProjectSpaces, it failed with the errors I noted above.

Plus, I cannot find any 'Install' button to install an OS into a ProjectSpace - there is only 'Replace' or 'Reinstall'.  So, I don't see how to Install a New OS into an Empty ProjectSpace successfully...

ZhuDaHai, 2019-08-24:
So, rereading the ReadMe again, this is what I think I understand:

1) Installing OSs from the Main Menu just puts them on the USB Drive outside of ProjectSpaces. This means that you cannot replace them or upgrade them when i newer version is available just four your RPi version (ie: Install ARCH2 on a Pi4B, and when ARCH4 is a available, it won't upgrade and you can't replace it)

2) ProjectSpaces basically only work with Rasbian or Debian OSes. rtandroid and Lakka is not compatible, so they can't be used there. So if you want to install either, you must install from the Main Menu, outside of ProjectSpaces. So if you install a pre-Pi4B version now you will not be able to upgrade it to a Pi4B version later - or add a Pi4B version later without wiping everything out and starting from scratch.

3) You basically install OSes into ProjectSpaces by selecting the OS, then going to the Maintenance menu and selecting the ProjectSpace, click on Replace and hope its compatible. There appears to be no way to reconfigure a ProjectSpace, even if its empty, to have a compatible number of partitions or be the proper size for a specific OS.

4) You can have multiple installs of the same OS, using ProjectSpaces, providing it is compatible with the ProjectSpace format.

So it looks like my best option is to just ditch the USB HD and buy a bunch of SD cards and manually install a separate OS on each one as PINN is just too restrictive.

ZhuDaHai, 2019-08-24:
Ok. Probably my last post. Running out of ideas and patience with PINN:

Why can't I partition the USB Drive so that is shows up as SDA1, SDA2, SDA3 and then just install/reinstall/wipe a single OS to each partition? 

I originally did this: partitions the USB Drive into 6 separate EXT4 partitions, and they showed up under Raspbian OS as devices: SDA1, SDA2, SDA3, etc. (30GB, 30GB, 30GB, 60GB, 60GB, 90GB) 
but when I used PINN, is just showed up as a single SDA device.

Seems to me this would have allowed me the full control that I'm needing.

procount, 2019-08-24:
Sorry that this is causing you so much frustration.
I don't have much time to respond at the moment so I'll just try to address your misunderstanding about projectspaces and OS partitions and try to address your other points later if they are then still relevent.

An OS has multiple partitions, normally a FAT32 partition for boot and an ext4 partition for the roots. So you cannot just create sda1 and install an os into it. Each os has a partitions.json file that tells pinn how to create these partitions.

Projectspaces just contain an empty FAT32 boot partition of about 100MB and an empty ext4 rootfs partition. 

Installing an os creates the partitions whereas replacing an os just replaces the files on the existing partition structure 

Rtandroid and lakka have 3 or 4 partitions so cannot replace a project space. They must be installed. Or you can create a special projectspace of your own with the same format as lakka or rtandroid and install that for later replacement.

PINN OSes are not image files but tar.xz files, so normally a new os has to be converted to this format before it can be used with PINN.

Installing OSes always wipes the disk format first. Hence the need for pojectspaces. This is a limitation from noobs which I have partially addressed with is replace and projectspaces. Currently it is not possible to reformat a projectspace for a different number of partitions due to the fact it changes all the partition numbers but I'm looking at ways to allow an os to be added later in a better way.

So to get lakka in your usb you need to install lakka and several projectspaces. My projectspaces can only be used for standard Linux 2 partition OSes.

Hope that helps for now.

ZhuDaHai, 2019-08-25:
Thanks!
I'm going to try using the USB Drive for NOOB based OSes and swap out the SD Card for other varieties for now.

closed 2019-08-25 by ZhuDaHai

bloodywinggithub, 2019-09-24:
兄弟你的octolapse分辨率问题解决了吗