procount/pinn #317: background.sh and pinn_init.sh not work!

giuspe01, 2019-09-16:
hi, i'm trying to make sure that when you enter recovery mode (via gpio) you start the raspbian system, i tried to use both background.sh and pinn_init.sh but none of the 2 works, i also noticed that pinn_init.sh it always runs even without entering recovery so I'm avoiding this method and I'm trying to use background.sh but it doesn't work my file contains only "rebootp 8", I also tried to run a second file but it doesn't work the same. Can anyone help me?

giuspe01, 2019-09-16:
or if it were possible to check whether the gpio 3 and 25 are connected and do a rebootp

lurch, 2019-09-16:
> hi, i'm trying to make sure that when you enter recovery mode (via gpio) you start the raspbian system

Could you explain in more detail what it is you're trying to do? If you want to _always_ start Raspbian, why not just do a standard image installation of Raspbian, without using PINN? I suspect I'm misunderstanding...

procount, 2019-09-17:
I think he is normally running another OS (like Retropie), but occasionally wants to run the 2nd OS (e.g. Raspbian) by asserting a GPIO on boot up.

Both of the scripts (background.sh and pinn_init.sh) will run independently of whether PINN has entered the recovery shell due to GPIO assertion or not. So you need a method of detecting the GPIO pin status,  I think you may need to add a gpio program to do that.

lurch, 2019-09-17:
You should be able to do it just using the `/sys/class/gpio/` stuff (e.g. using bash)?
I have no need for it myself, but I guess it'd be super-cool if you could associate a different GPIO pin with each of the boot-OS menu choices in PINN :wink:

giuspe01, 2019-09-17:
I solved by checking the pinns at startup co pinn_init.sh

thank you very much for helping.

closed 2019-09-17 by giuspe01