procount/pinn #318: Possibility to give a command to reboot into the other non-default OS when having 2 OSes installed

garret, 2019-09-16:
I installed 2 operating systems thanks to PINN: DietPi and Batocera for retrogaming in a 32GB card on a raspberry pi 3b+. I have set up DietPi to be the default OS. 

Would it be possible to give a command in DietPi to reboot the raspberry pi and boot to Batocera? 
(but still keeping DietPi as default OS every time the raspberry pi is restarted again)

procount, 2019-09-17:
Depending on whether your OSes support it, you could try:
`reboot n`
where n is the partition number of the boot partition of the OS you want to boot (typically 6 or 8).

lurch, 2019-09-17:
@procount I wonder if it's worth including a statically-linked version of your rebootp program on the recovery partition, so "all" a user would need to do would be something like:
```bash
mount /dev/mmcblk0p1 /mnt
/mnt/rebootp 6
```
? (although perhaps it wouldn't really be that easy :man_shrugging: )

garret, 2019-09-17:
Thank you for your messages but I admit I am quite noob with PINN. I cannot try right now the command as I don't have access to the raspberry pi. I will do it later today when coming home. 

How do I find out which number is the OS I would like to boot into? Is there a command I can give to show a list? 

What if the OS I am using (DietPi) does not support this function. Is there a way to make it support this function eventually?

procount, 2019-09-17:
> How do I find out which number is the OS I would like to boot into? Is there a command I can give to show a list?

From PINN's menu, go to the recovery shell by pressing ctrl-alt-F2 and login with root/raspberrypi
`less /settings/installed_os.json`
This will show a list of the OSes that are installed. Each OS will contain a lot of information, including the partition numbers where it is located such as `/dev/mmcblk0p6` The last number of the first partition for each OS is the partition number you want to use.

> What if the OS I am using (DietPi) does not support this function. Is there a way to make it support this function eventually?

I think @lurch's suggestion would work just fine.
Also I think @matthuisman already has some plugins and programs to do this for variosu dual-boot OSes.

garret, 2019-09-17:
@procount the command `reboot n` on DietPi works like a charm and reboots into Batocera. On the other hand, `reboot 8` does not work in Batocera. 

Therefore, I tried to give in Batocera these commands as suggested by @lurch but without success: 
```
# mount /dev/mmcblk0p1 /mnt
# /mnt/rebootp 8
-sh: /mnt/rebootp: No such file or directory
# /mnt/reboot 8
-sh: /mnt/reboot: No such file or directory
```

Do you know how I can fix this? I also gave a `ls` command in `/dev/`, maybe it can help? 
```
# ls /dev/
autofs              loop0               ram0                shm                 tty24               tty45               tty9
block               loop1               ram1                snd                 tty25               tty46               ttyAMA0
bsg                 loop2               ram10               stderr              tty26               tty47               ttyprintk
bus                 loop3               ram11               stdin               tty27               tty48               uinput
cachefiles          loop4               ram12               stdout              tty28               tty49               urandom
char                loop5               ram13               tty                 tty29               tty5                vc-mem
console             loop6               ram14               tty0                tty3                tty50               vchiq
cpu_dma_latency     loop7               ram15               tty1                tty30               tty51               vcio
disk                mem                 ram2                tty10               tty31               tty52               vcs
fb0                 memory_bandwidth    ram3                tty11               tty32               tty53               vcs1
fd                  mmcblk0             ram4                tty12               tty33               tty54               vcs2
full                mmcblk0p1           ram5                tty13               tty34               tty55               vcs3
gpiochip0           mmcblk0p2           ram6                tty14               tty35               tty56               vcs4
gpiochip1           mmcblk0p5           ram7                tty15               tty36               tty57               vcsa
gpiomem             mmcblk0p6           ram8                tty16               tty37               tty58               vcsa1
hidraw0             mmcblk0p7           ram9                tty17               tty38               tty59               vcsa2
hidraw1             mmcblk0p8           random              tty18               tty39               tty6                vcsa3
hidraw2             mmcblk0p9           raw                 tty19               tty4                tty60               vcsa4
hwrng               network_latency     rfkill              tty2                tty40               tty61               vcsm
input               network_throughput  sda                 tty20               tty41               tty62               watchdog
kmsg                null                sda1                tty21               tty42               tty63               watchdog0
log                 ptmx                sdb                 tty22               tty43               tty7                zero
loop-control        pts                 sdb1                tty23               tty44               tty8

```

procount, 2019-09-17:
Lurch's solution requires me to do a bit of work first to make rebootp available. Currently it is embedded in recovery.rfs.

garret, 2019-09-17:
Ok, so it means that I have to wait until a new release of PINN is released?

procount, 2019-09-18:
Yes. I'll take a look though.

garret, 2019-09-18:
@procount Would that mean I have to format all the sdcard and thus "lose" the two oses installed or there will be a way to update PINN with no implications? Sorry but I just started with PINN and I am quite noob

procount, 2019-09-18:
No.
But anyway, there is a way you can do it now (since rebootp is already statically linked :wink: )

Boot into PINN (press shift to enter the recovery menu if it doesn't do it itself).
Press ctrl-alt-F2 and login to the recovery shell with root/raspberry.
Type the following:
```
mount -o remount,rw /mnt
cp /usr/bin/rebootp /mnt
sync
```
Press ctrl-alt-F1 to go back to the recovery menu.
Boot into Batocera (or other OS).
In Batocera command shell, type the following:
```
mount /dev/mmcblk0p1 /mnt
sudo /mnt/rebootp 8
```
Of course, alternatively, you could just copy rebootp into the /usr/bin folder of any OS where the reboot does not already work.

garret, 2019-09-18:
@procount Thank you very much! I confirm that it works perfectly :)

Cyberoliv, 2020-05-23:
Hi, 

I'm trying to do the same thing to reboot from **raspbian** to batocera.

My default sticky partition (auto-boot) is raspbian.
My batocera partition is mmcblk0p8.

I tried `sudo reboot 8 `
=> Reboots to raspbian

So I tried to copy rebootp to /usr/bin as suggested above

Permissions ok... go !
`sudo rebootp 8 `

reboots, but still on raspbian default partition :( 

Am I missing something ?

procount, 2020-05-23:
Which model of Pi?

Cyberoliv, 2020-05-23:
Raspberry pi 4

procount, 2020-05-23:
See https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=257845&p=1664625#p1664625

Cyberoliv, 2020-05-24:
Works like a charm thank you ! 👍 

I read performance drop of the SD card that I saw, I will follow the thread to see if it improves.

garret, 2022-06-13:
I am re-opening the issue because I cannot get `rebootp` to work on a rpi4. 

I have PINN installed on a SDcard with two OSes: Batocera and raspberrypiOS (64bit). At the moment, I am trying to get to reboot from raspberrypiOS to Batocera. 

That does not work with the classic: 

```
sudo reboot 6
```

And I guess it is because I have to edit the `cmdline.txt` as also written above. However, I did not want to recur to such solution as I am worried about the SDcard performance drop. 

Therefore, I copied `rebootp` from PINN to `/mnt` as described in a previous message and executed it from raspberrypiOS. However, it seems to do not work: 

```
garret@raspberry:~ $ sudo mount /dev/mmcblk0p1 /mnt
mount: /mnt: /dev/mmcblk0p1 already mounted on /mnt.
garret@raspberry:~ $ sudo /mnt/rebootp
sudo: unable to execute /mnt/rebootp: No such file or directory
```

```
garret@raspberry:~ $ ls /mnt/
 bcm2708-rpi-b.dtb          config.txt
 bcm2708-rpi-b-plus.dtb     defaults
 bcm2708-rpi-b-rev1.dtb     firmware.tar.gz
 bcm2708-rpi-cm.dtb         fixup4rc.dat
 bcm2708-rpi-zero.dtb       fixup_rc.dat
 bcm2708-rpi-zero-w.dtb     INSTRUCTIONS-README.txt
 bcm2709-rpi-2-b.dtb        os
 bcm2710-rpi-2-b.dtb        overlays
 bcm2710-rpi-3-b.dtb        overrides.json
 bcm2710-rpi-3-b-plus.dtb   rebootp
 bcm2710-rpi-cm3.dtb        recover4.elf
 bcm2710-rpi-zero-2.dtb     recovery7.img
 bcm2710-rpi-zero-2-w.dtb   recovery7l.img
 bcm2711-rpi-400.dtb        recovery.cmdline
 bcm2711-rpi-4-b.dtb        recovery.cmdline.new
 bcm2711-rpi-cm4.dtb        recovery.elf
 bcm2711-rpi-cm4s.dtb       RECOVERY_FILES_DO_NOT_EDIT
 bootcode.bin               recovery.img
 BUILD-DATA                 recovery.rfs
 changefirmware             riscos-boot.bin
 config.dsi                 ssh
 config.hdmi               'System Volume Information'
 config.hyp3                wallpaper.jpg
 config.hyp4
garret@raspberry:~ $
```

Could it be because I am running raspberrypiOS as 64bit OS while rebootp is for 32bit systems? In any case, do you know how to solve such issue?

procount, 2022-06-13:
This approach will not work, on 2 counts.

If `sudo reboot 6` does not boot into the correct partition, then `sudo rebootp 6` will not work either.
This is due to the hardware on the Pi4, specifically the PMIC. To workaround it, you MUST apply the cmdline quirks to each OS you want to reboot from.
Have you actually tried it? The performance drop may not even be noticeable in real world use. If it is not suitable for your needs, you can always revert the change. It is not permanent and does no harm. Do not be worried.

Your method of copying rebootp will not work , since it is in the recovery.rfs file of PINN. You must boot into PINN's environment, mount the partition of the OS and copy rebootp from PINN to the OS. But as I said, this is probably fruitless.

garret, 2022-06-13:
I tried with the `sudo reboot 6` after editing the `cmdline.txt` with the quirk and it worked, so I will close again this issue.

closed 2022-06-13 by garret