procount/pinn #319: ubuntu not available!

DhrubojyotiDey, 2019-09-22:
can't find ubuntu. please fix this.

procount, 2019-09-24:
Which one?
Raspex is based on Ubuntu for RPi 2 & 3
Raspex4 is based on Ubuntu for Rpi 2,3 & 4
Ubuntu-mate 16.04 is available for Rpi 2 &3
I was hoping for a new Ubuntu-Mate 18.04, but it has not materialised yet.
Which Model Pi do you use?

DhrubojyotiDey, 2019-09-24:
Well I use the Raspberry pi 4B. So I was wondering if I could get the
Ubuntu up and running there. Another thing is that how do I boot multiple
OS ? I'm looking for installing the Ubuntu, KODI, and android. That's all.
Really appreciate your work on this. Thanks for helping me out with this!

procount, 2019-09-24:
So, the only Ubuntu based OS I have for the Pi4B is Raspex4 at the moment.
For Kodi, you can select LibreELEC.
I don't think android is available yet for the Pi4.
To install Multiple OS, just select the ones you want and install them together. Maybe select a ProjectSpace as well to reserve some space for you to install another OS into later, because you cannot ADD an OS at a later date, only REPLACE them (if they are compatible). The alternative is to backup each OS, and reinstall them with any additional OSes you want. Possible, but it takes a while. :)

RaspberryPiFan, 2019-10-12:
> I don't think android is available yet for the Pi4.
There isn't an build, but if someone is interested, he can use this repository: https://github.com/RaspberryPiFan/local_manifests

procount, 2020-11-08:
Ubuntu 2010 and Ubuntu Mate 2010 are now available

closed 2020-11-08 by procount

lurch, 2020-11-08:
I thought "2010? That was 10 years ago! Surely you mean Ubuntu 2020?"  And then I was just about to write that, when I realised you meant Ubuntu 20.10 :rofl: :rofl: