procount/pinn #321: Issue with debugging pinn: Recovery shell doesn't supports other keyboards

RaspberryPiFan, 2019-10-14:
If you press ctrl-alt-f2 on an running PINN system and try to login and have a QWERTZ keybord, it uses QWERTY layout even if you chose QWERTZ in PINN.

RaspberryPiFan, 2019-10-14:
I hopefully can fix this with new buildroot.

lurch, 2019-10-14:
Interesting. I suspect the recovery shell is using the keyboard layout chosen by the Linux kernel / console / bash / busybox setup, and changing the keyboard in "PINN" changes the keymap at the Qt (GUI) layer, but doesn't change any of the lower layers.

RaspberryPiFan, 2019-10-15:
Yes, it also could be fixable by removing the "purge unwanted locales" option in buildroot.

RaspberryPiFan, 2019-10-15:
If it's not fixable, it would possible to add a warining to README or change the default password